vitormhenrique/OctoPrint-Enclosure #78: Relay turns on but not off.

JD2005L, 2017-12-13:
If I clear the GPIO mode, the relay turns off, then I can turn it on once, but when trying to turn it off I get the following error:

`2017-12-13 14:31:22,520 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on updateOutputUI. Arguments:
('Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)',)
`

vitormhenrique, 2017-12-13:
You should not be clearing the GPIO modes, this is intended for advanced use cases when you have incompatibility between two plugins.

After clearing, you need to change the mode, and save the settings, if you don't you will experience problems.

Best regards,
Vitor Henrique

vitormhenrique, 2017-12-18:
Did it fix your issue? Let me know so I can close this thread.

JD2005L, 2017-12-18:
I was only clearing the GPIO mode because it wasn't working in the first
place.

vitormhenrique, 2017-12-18:
Bud did you got it working?

vitormhenrique, 2017-12-18:
What board are you using? what is the configuration that you are using? We need more information to help you.

JD2005L, 2017-12-18:
Had to replace the pii was using for whatever reason.

vitormhenrique, 2017-12-18:
Oukay, I'm going to close this issue.

closed 2017-12-18 by vitormhenrique