vitormhenrique/OctoPrint-Enclosure #80: Temperature Control - Relay turn on/off

dismaster, 2017-12-18:
Hi,
When setting the temperature control to manage fans it would be good if temperature is back below the configured to turn off the relay. Is this possible?

Greets

vitormhenrique, 2017-12-18:
Sorry, I did not quite understand your request. What do you mean by "back below the configured to turn off the relay"?

dismaster, 2017-12-18:
Lets say I’m configuring to turn GPIO pin xx when temperature of 30 degrees have been reached to cool down the system. When it cools down let’s say to 28 degrees I would like to toggle GPIO pin xx. Meaning it’s automatically switches fans on on high temperature and off on low temperature.

vitormhenrique, 2017-12-18:
but you can already do this. 

just use the temperature control, select cooling, instead of the GPIO control. It will track the temperature and turn on and off the fan whenever it needs.

dismaster, 2017-12-18:
Thanks - please close, sorry for inconvenience.

vitormhenrique, 2017-12-19:
no problem...

closed 2017-12-19 by vitormhenrique