vitormhenrique/OctoPrint-Enclosure #82: OctoPrint-Enclosure

brwab, 2018-01-02:
Hi,

i really like your Enclosure plug in. Its grate.

Unfortunatly i can not get the Sensors to work. I Have a GY-BME280-3.3 High Precision Atmospheric Pressure Sensor Module Link: (https://www.amazon.de/GY-BME280-3-3-Precision-Atmospheric-Pressure-Sensor/dp/B01LWL13TP)
and also a HTU21D but i can not get them to work with the plugin.

Can you please lead the way to get them working?

Thank You.

vitormhenrique, 2018-01-02:
the plugin documentation clearly especify the sensors that are supported (DHT11, DHT22, AM2302, DS18B20, SI7021, BME280 and TMP102)

https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/README.md

I can not add support to every sensor out there, especially if I don't have the sensor. And I like to test the sensor when I add it. So you hate few options:

- implement the sensor following the examples on my plugin.

- ship me a sensor / lend me a sensor /  so I can add it and test it

- use one of the supported sensors, there are a bunch supported, most are inexpensive, please note that I do not recommend using dht sensors.

vitormhenrique, 2018-01-02:
Sorry, i just noticed that bme280 is the same sensor that you are talking about. 

You need to get the sensor working first, you need to configure your board to use I2C instead of SPI.

### Configure the hardware:

**VIN to	3V3
GND	to	GND
SCL	Clock to	GPIO 3 (SCL)
SDA	Data to	GPIO 2 (SDA)**

### Configure the software:

ssh on your rpi and follow the commands:

1. Run sudo raspi-config
2. Use the down arrow to select 9 Advanced Options
3. Arrow down to A7 I2C
4. Select yes when it asks you to enable I2C
5. Also select yes when it asks about automatically loading the kernel module
6. Use the right arrow to select the <Finish> button
7. Select yes when it asks to reboot

8.Install some packages:
`sudo apt-get install i2c-tools python-pip` 

9.Find the address of the sensor:
`i2cdetect -y 1`

10.It should display something like:
<pre><code>0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
  00:          -- -- -- -- -- -- -- -- -- -- -- -- --
  10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
  70: -- -- -- -- -- -- 76 -- </code></pre>

11. download a test script:
`wget https://gist.github.com/vitormhenrique/2c4617353992666359b5493c1d684321`

12. run the script using:
`python [NAME OF THE FILE] [address of the sensor]`

So if you name the script "BME280.py" and the address is 76 you should run:
`python BME280.py 76`

if you get the sensor working, configure the settings with the address and you should be good to go.

brwab, 2018-01-14:
Hello Vitor,

thank you for your help, it took me some time but i got it working now (was to busy lately).

But now that it is working i have another question.

It seams to me that there is ether the option to heat or to cool the enclosure. Is it possible to have the plugin heating when its too cold and cooling when its too hot just by setting the desired temperature?

And also id like to be able to read a second Temperature Sensor to control the water cooling i have build in my printer (control radiator fans with radiator temp sensor) to keep it quite when printing at low temps.

I really appreciate your help and the work you have done with the Plugin.

Thank You very much.

 
Bruno Wabnitz Hofgut Schwarzenborn 1 36132 Eiterfeld Germany Tel.: +49 (0)6672-86 79 099 Mob.: +49 (0)151-16 30 74 74 

 
 

   
   - Run sudo raspi-config
   - Arrow down to A7 I2C
   - Use the right arrow to select the button

vitormhenrique, 2018-01-15:
I guess I could use the first reading temperature when setting the temperature to figure it out if we want t a heater or a cooler, multiple persons requested multiple heaters. I'll add a feature request.

closed 2018-01-15 by vitormhenrique