vitormhenrique/OctoPrint-Enclosure #83: Feature request: multiple heater support

zbrozek, 2018-01-02:
It'd be lovely to have native support for multiple heaters on several GPIOs. In my case, for example, I have a hair dryer with two heating coils. Right now, octoprint-enclosure controls one of them and I added an additional GPIO so that I could control it manually. I use it to speed up enclosure heating when getting started, but it's overkill during steady state.

Even switching multiple GPIOs together for a single heater would be really nice.

vitormhenrique, 2018-01-02:
You might be able to do this already.

You can create one output with auto startup for coil number 2 and an input temperature that will shut it down after it gets to a set temperature.

so whenever a print starts, you will have this bust on the temperature until the enclosure temperature get's to a defined temperature, it will later shut it down, and keep coil number 1 normally to heat the enclosure whenever needed.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

zbrozek, 2018-03-03:
I just installed the dev version of the plugin. It looks really nice in general - it really shows that you've put a lot of work into it!

Is there a log file somewhere to see what's going on? It seems like my previously-working DS18B20 sensor isn't working. Or something else is going on to prevent temperature readout.

vitormhenrique, 2018-03-03:
Now you need to put the serial for DS18B20 sensors. Have you done so?

Best regards,
Vitor Henrique

zbrozek, 2018-03-03:
I saw that, but assumed I didn't need to do anything because I had only one sensor. Upon going back to edit settings, it looks like changes have no effect and the layout of the page has changed. Seems like something broke itself? I'm uninstalling the plugin and reinstalling it to see if my incorrect sensor setup blew something up.

zbrozek, 2018-03-03:
OK, that was super weird. Upon reinstall the UI returned to new-normal-looking and recovered my settings. Adding the slave address (in my case, 031722d7a8ff ) didn't cause a temperature to show up. Do I need to be connected to the printer or printing to show the temperature?

vitormhenrique, 2018-03-03:
031722d7a8ff  does not look like a DS18B20 serial, go to this folder:

`cd /sys/bus/w1/devices`

And type 

`ls`

You should probably see something starting with 28-, that is the sensor serial number.

zbrozek, 2018-03-03:
Ah, the 28-xxxx led me to think only the xxxx was the desired serial number. I don't usually think of "-" as hex, so that cemented my assumption. Putting in the whole string 28-031722d7a8ff made sensing work. Dual heating element control also (at first glance) seems to work. Thanks!

I'll keep testing it and report back if anything goes wrong.

zbrozek, 2018-03-06:
So far so good. I've done a bunch of prints and this seems to be working nicely. I set one heater to target a lower temperature than the other and it speeds up initial warmup considerably. Fantastic!

Thanks so much!

closed 2018-03-06 by zbrozek

vitormhenrique, 2018-03-06:
You can either do that, or use a temperature alarm and disable one output when your set temperature is reached.