vitormhenrique/OctoPrint-Enclosure #84: Error ('The GPIO channel has not been set up as an OUTPUT',) when attempting to turn on pin

putertdi, 2018-01-07:
I've got everything set up and wired but I can't seem to get the output voltage on the pins. Every time I try to turn on the heater I don't see any voltage change and I get this error in the log:

2018-01-06 23:41:37,253 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on writeGPIO. Arguments:
('The GPIO channel has not been set up as an OUTPUT',)

I've verified that I'm not seeing voltage between gpio pin 23 or pin 24 to ground when I turn them on (to rule out a bad GPIO pin).  Heres a screenshot of my configuration:
[image]

Can you give any advice on how I should fix this? I've tried the following:
1) octoprint service restart
2) raspberry pi restart
3) clear GPIO button on configuration
4) remove and re-add all of the ports.
5) checked for octoprint updates

I'm at a bit of a loss on what to do next.

Thanks!

vitormhenrique, 2018-01-07:
Hello, On the bottom of the page what mode is gpio set to? BCM or BOARD?

Check and uncheck the check box "Use Board Pin #" after you press the clear GPIO mode.

Save, re enter settings delete all outputs, save, enter settings again, create the GPIO output again.
If you can give me access to your pi I can try trouble shoot.

What PI version are you using? Also, are you using your own octoprint installation or octopi?

Some people had some sort of incompatibility on their system and they needed to re-instal octoprint.

vitormhenrique, 2018-01-07:
Check issue 78, it looks similar.

putertdi, 2018-01-07:
Thanks for the quick reply.  the "Use Board Pin #" under advanced is unchecked.  

When I try to mark and unmark the checkbox I get this error when marking it after I hit save:
"GPIO mode was configured before, GPIO mode will be forced to use: BCM as pin numbers. Please update GPIO accordingly!"

I tried the steps you gave and I'm still getting the same error.

I have a raspberry pi 1 revision 1.2.  I THINK it's a B+.  I'm using an octopi install - though It's an old one and I've long since updated octoprint etc so I'm not sure how important that is.

I'd be happy to give you access to the raspi if you think it would help - do you have a good private communication method where I can give you login info etc?

here's a link to a screenshot of the checkbox: 
[image]

putertdi, 2018-01-07:
I don't think issue 78 is the same.  I can't turn on the port at all and I get a different error.

putertdi, 2018-01-07:
So, I wiped the settings and tried setting up again and I think I found the issue.  When I go in and add the new output (don't change any other settings) and hit save I get this error:

2018-01-07 03:19:23,841 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on configureGPIO. Arguments:
('No access to /dev/mem.  Try running as root!',)
2018-01-07 03:19:31,327 - octoprint.plugins.enclosure - WARNING - An exception of type RuntimeError occurred on updateOutputUI. Arguments:
('You must setup() the GPIO channel first',)

I think it's failing to set the pin state and that's why I'm getting the error later on but I have no idea how to go about solving this.

Any suggestions?

vitormhenrique, 2018-01-07:
it's probably because of the old install,
You can try to fix the access to the GPIO from your user

`sudo adduser pi gpio`

then:

`sudo chown root.gpio /dev/gpiomem`
and
`sudo chmod g+rw /dev/gpiomem`

Check [this](https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root) for more information.

If you don't want to mess around with linux to much, just re-flash octopi with the latest image, it should be fine..

putertdi, 2018-01-07:
Unfortunately, I already tried that article :(

I have a lot of other configuration done at this point so I want to avoid reflashing.  I'm currently pulling a backup of the card and then I'm going to upgrade the os.

One thing to note that is that gpiomem doesn't exist for me.  I tried the same comments on /dev/mem.

vitormhenrique, 2018-01-07:
let's try to first figure it out if the error is with the pi access to GPIO.
Just create a simple python script like:
```python
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BCM)
GPIO.setup(18,GPIO.OUT)
print "LED on"
GPIO.output(18,GPIO.HIGH)
time.sleep(1)
print "LED off"
GPIO.output(18,GPIO.LOW)
```

Name it _**test.py**_, then try to run without sudo:

python ./test.py

See if you have any error....

putertdi, 2018-01-07:
good idea.  I'll do this as soon as the backup finishes and I can boot the pi.

vitormhenrique, 2018-01-07:
It's time for me to hit the bed now, I'll keep trying to help you tomorrow if it doesn't work after upgrading...

putertdi, 2018-01-07:
Sounds good - have a good night.

Thank you again for all the help.

putertdi, 2018-01-07:
Ok, so I've got this 99% fixed, but could use your input on the last 1%.  Also note that I will edit my original comment to make it more searchable for symptoms of this issue and have a clear step by step set of instructions for fixing it for anyone in the future (and so that you can reference this if/when you run into this with others as this is not something that I think will be specific to me...in fact you said some people had to reinstall octoprint and I think that was just a more aggressive way of solving this).

There are two issues going on.  First issue is that my OS was too old - it didn't have the /dev/gpiomem implementation intended to protect /dev/mem from over permissions while allowing access to gpio.  Updating the firmware and OS fixed this part.

The second part is lack of access to /dev/gpiomem.  /dev/gpiomem comes in with these permissions:
$ ls -l /dev/gpiomem
crw------- 1 root root 244, 0 Jan  1  1970 /dev/gpiomem

But it needs these:
$ ls -l /dev/gpiomem
crw-rw---- 1 root gpio 244, 0 Dec 28 22:51 /dev/gpiomem

When I chown and chmod those permissions in using the below everything starts working:
sudo chown root.gpio /dev/gpiomem && sudo chmod g+rw /dev/gpiomem

Here's the rub though, as soon as you reboot those permissions get blown away and everything stops working again with the same set of errors.

Here's where I need your help.  I THINK I can fix that using information on this page: 
https://raspberrypi.stackexchange.com/questions/60240/what-sets-ownership-and-privileges-on-dev-gpiomem

My concern is that I _shouldn't_ have to do this.  This plugin apparently works for everyone else.  So while I can force stuff in place to have the permissions needed to do this, I'm worried that there's a different configuration symptom/problem.  Can you offer any guidance on what may be going on here?  My hope is that you've seen something like this or have some idea on what may be missing.

putertdi, 2018-01-08:
As a followup, I can confirm this works if I manually set the permissions (though I need to restart the service or remove and re-add the outputs to get everything working).  I'm still holding off on applying the workaround to update the permissions every boot in hopes that I can find the "correct" solution.

vitormhenrique, 2018-01-09:
Hello, sorry for the delay on the response.

Do you have the same symptoms when just running the python code that access the GPIO? or just running octoprint?

Have you double checked which user is running the octoprint server? Do you have a spare sd card, just flash a new octopi, install and do a quick test with a fresh installation, that should take ~1 hour, and just to sanity check.

putertdi, 2018-01-09:
I do have the same symptoms.  This is _definitely_ a permissions issue.  The problem is that the permissions get reset every reboot.

When I boot, you'll see this output and you won't be able to write to gpio without being root:
```
$ ls -l /dev/gpiomem
crw------- 1 root root 244, 0 Jan  1  1970 /dev/gpiomem
```

If I run this command, everything will start working:
`sudo chown root.gpio /dev/gpiomem && sudo chmod g+rw /dev/gpiomem`

And the output will end up looking like this:
```
$ ls -l /dev/gpiomem
crw-rw---- 1 root gpio 244, 0 Dec 28 22:51 /dev/gpiomem
```

The problem here is that everything gets reset.  If you look above I have a link to a workaround (editing some startup scripts to update the permissions back to what they should be) but I'm wondering why I have to do this and no one else does.

vitormhenrique, 2018-01-15:
I have no clue, maybe to old image that you started?
I'll close this issue, and thank you for documenting your findings. This might help people having the same issue. :+1:

closed 2018-01-15 by vitormhenrique