vitormhenrique/OctoPrint-Enclosure #85: API Calls to trigger GPIO IO Pots

Modding, 2018-01-11:
Hi,

i really love your Plugin, and i am also using printoid to manage my Printers (Anet A9 and Prusa I3 Steel).

And i want to switch the Power of the Relais on and off with printoid. For this Task i need proper API Calls that work. 

Example:
curl -s -H "Content-Type: application/json" -H "X-Api-Key:
APIKEY" -X POST -d '{ "command":"IO1ON" }' --user MyHTACCESSUSER:MyHTACCESSPassword http://MYOCTOPRINT/api/plugin/enclosure

Is this feature availible but not documented or is it a feature Request?

Greetings from Germany 

Valentin

vitormhenrique, 2018-01-11:
It's definitely a feature request. 
It's possible to implement and not that difficult, I need to think the best way to implement the API and make it generic and expandable.

http://docs.octoprint.org/en/master/plugins/mixins.html#simpleapiplugin

I'll try to include on next release.

I'm really busy with some other projects so I have no ETA on this,

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-01:
To set GPIO status you need to send a request to

`/plugin/enclosure/setIO?status=false&index_id=1&apikey=**{APIKEY}**`

Note that index_id is the number under settings for that output, and status is true to turn it on and false to turn the output off.

You also need to add the apikey on the request, it will be under FEATURES settings for octoprint.

You can request current gpio status with

vitormhenrique, 2018-03-05:
API documented here https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/API-Control

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique