vitormhenrique/OctoPrint-Enclosure #87: Heater shutdown occurs at end of print even when not enabled.

putertdi, 2018-01-18:
The plugin is consistently turning off the heater at the end of the next print if I set a target temp.  This is despite me not having the "auto shutdown" checkbox unmarked.

vitormhenrique, 2018-01-18:
Yeap.. I think I was disabling the heater at the end by default. 

I'm starting to add more functionality on this plugin now and will be fixing those issues.

vitormhenrique, 2018-01-22:
I'm going to fix this on the next release, but I just noticed that the temperature control does not have an auto shutdown field, are you using the temperature control or the gpio output?

putertdi, 2018-01-24:
sorry for the delay, I was sick the last couple days.

I was referring to the "auto shutdown" checkbox on the gpio pin.  I don't see a choice present for the enclosure temp settings.

As I use the plugin more - I'm realizing sometimes you want it to turn off after a print and sometimes you don't.  I'd recommend adding the checkbox next to the enclosure tab options rather than on the config.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

cmgerdes, 2018-03-06:
Sorry for the lack of response.

I've been swamped and haven't had a chance to load up the new version.  I'll let you know as soon as I do.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique

cmgerdes, 2018-03-07:
well then - that makes installing it easier ;)

Honestly, I haven't done any printing for over a week because I've been busy with other stuff.  Thank you for doing this!