vitormhenrique/OctoPrint-Enclosure #88: Minor issue: Donate/PayPal butto

Petclaud, 2018-01-21:
Tried to make a small donation, but the Donate/PayPal button does not link on gifthub :-)

Maybe the wrong place - sorry - but a kind request for some help: 

I tried to change the TAB order in OctoPrint. The config.yaml file was not present in the folder ~/.OctoPrint hence I created the file via sudo nano config.yaml. In the editor I filled in the following:
appearance:
  components:
    order:
      tab:
      - temperature
      - control
      - gcodeviewer
      - terminal
      - plugin_enclosure
      - timelapse

However, the TAB order is still the same. Any thoughts/advice..?

vitormhenrique, 2018-01-22:
Hello! thanks for letting me know about the button, I think I fixed...

Are you using an octopi installation or have you installed octoprint from scratch?

vitormhenrique, 2018-01-22:
btw,  I'm adding a bunch of new features and minor fixes! new release is coming soon...

vitormhenrique, 2018-01-26:
Hello, did you get the tab order thing worked out?

Petclaud, 2018-01-27:
Hello. Thx for your feedback. No I did not get the tab order changed via editing the config.yaml file. I do have on one printer an almost 'clean and fresh' Octopi installation and on another one that has been updated already several times (has been running for more than a year). On both of them I could not make it work. However.... a new plugin has arrived (https://plugins.octoprint.org/plugins/taborder/) and that did it for me. Thx for your support!

vitormhenrique, 2018-01-31:
Awesome! closing this issue! :)

closed 2018-01-31 by vitormhenrique