vitormhenrique/OctoPrint-Enclosure #89: Help wanted: It does not work.... turning on or off an output has no effect.

TGC-TECH, 2018-01-25:
I am using BCM pin 14 (board pin 8) no matter if  I enter it as  8 or 14 in the settings it does not work. 
I am uploading my logs, any help would be appreciated. 
Thanks! 
and sorry for the stupid question 
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1665657/octoprint.log)

vitormhenrique, 2018-01-25:
What is the model of you raspberry pi?
Do the following steps:

1. delete all your configurations, save and close the setting screen.
2. go to the advanced part of the plugin settings and click the button "Clear GPIO Mode", close setting screen again, and restart the server
3. reopen settings and add only add one output just for test

Another good idea is  to first figure it out if the error is with the pi access to GPIO.
Just create a simple python script like:
```python
import RPi.GPIO as GPIO
import time
GPIO.setmode(GPIO.BCM)
GPIO.setup(14,GPIO.OUT)
print "LED on"
GPIO.output(18,GPIO.HIGH)
time.sleep(1)
print "LED off"
GPIO.output(14,GPIO.LOW)
```

Name it _**test.py**_, then try to run without sudo:

`python ./test.py`

See if you have any error....

vitormhenrique, 2018-01-25:
I think you are having the same issue of 84.

Are you using an old octopi image? those have issues with access to the GPIO using pi user. 

You can check if that is the case if you try to create the script that I told you and run as the pi user. If you get errors than this is the same issue.

I suggest download a new image, and install the plugin again if that is the case.

vitormhenrique, 2018-01-25:
By the way, the log that you sent GPIO was set to be using BOARD numbers....

TGC-TECH, 2018-01-26:
I know it was using board numbers which is why I tried 8 originally. 
The Python script worked once I changed the 18 to 14... and I have now made 2 python scripts, one to turn the printer on and one to turn it off and I am using the system command editor plugin to add buttons to run the python scripts and the GPIO, (It is all working now) thanks!

vitormhenrique, 2018-01-26:
no problem...

closed 2018-01-26 by vitormhenrique