vitormhenrique/OctoPrint-Enclosure #90: serial number Starts with 10-*

MKamina, 2018-01-26:
hi there the serial number of my DS18B20 start with 10-* so the plugin don't show up the Temp in nav bar. where can i chang it? 
thanks for help.

vitormhenrique, 2018-01-26:
I'm going to change the setting screen and allow user to select the serial number for each sensor, I'm adding support for multiple sensors, so that will be a most,

For now you can edit the this line https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/dev/octoprint_enclosure/__init__.py#L341

If you have a default octopi image the file should be at:

/home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/__init__.py

MKamina, 2018-01-26:
Thanks! helped me much!

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique