vitormhenrique/OctoPrint-Enclosure #92: Filament sensor switch not working at all

fdhosting, 2018-02-08:
Hello !
I have installed your nice plugin, and want to use it for a fillament sensor.
I have octoprint running on a RPI3 and I have connected a switch (see attachment) to GPIO26 (pin 37) and GND (pin 39).

SETUP:

Raspberry Pi Inputs
EVENT: printer
Input IO Number: 26
Input Pull Resistor: INPUT PULL UP
Event Trigger: FALL
Printer Action: fillament change
Filament Detection Timeout: 120
Use Board Pin #: not selected

When I start printing and press the switch during a few minutes: nothing happens. If I release the switch, nothing happens also..

I have this code in the fillament change codeblock:

G199 ; pauses the print immediately, and moves the head to X0, Y100.

Don't know what's the problem, doesn't work at all..

Can you help ? I think a lot off people wants to use your plugin..
Thanks in advance !

;-) Arthur

PS: I have also connected an DT22 to GPIO4 (pin 7), and installed the software library for it: that works, I see the temperature and the humity.

[image]
[image]

vitormhenrique, 2018-02-08:
I need a couple of things:

1. What pins have you used to connect the switch? there are three pins, C, NC and NO.

2. Have you tested a simple input button python script to test your wiring? 

ssh on your pi, create the a new file

```python
import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)

GPIO.setup(26, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    input_state = GPIO.input(26)
    if input_state == False:
        print('Button Pressed')
        time.sleep(0.2)
```

Name it _**test.py**_, then try to run without sudo:

`python ./test.py`

Press the switch, if nothing happens that means that you have something wrong with your wiring... 

3. If the script works, send me your octoprint.log

fdhosting, 2018-02-09:
the script works, I get the message 'Button Pressed' when I press the button.

Printer doesn't respond to the button at all when printing.

I have added the log..

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1711570/octoprint.log)

fdhosting, 2018-02-09:
The pause command for the Craftbot Plus printer is: G197 and the resume command is G198.

It would be nice if the printer could go to pause mode after pressing the button (later it will be replaced by a fillament-sensor switch). Also the extruder head must be parked to the left side.

After the resume command, the extruderhead must be moved to the last printerposition and continue with the printerjob.

Until now, it doesn't work at all.
Strange thing: the DT22 works nice...

vitormhenrique, 2018-02-09:
Something is a little bit odd...

despite you mention that use board number is not marked, the log is telling that it is being used.

What happens if you change the filament type action to pause? does it work? 

Reboot your pi, it will clear gpio modes, change the setting to pause /resume, and test it out, and send me the log again.

fdhosting, 2018-02-09:
Ok, rebooted again. Checked if the board number is marked: it's not marked.
Printing again, pressed the switch a few times, nothing happens at all ??

[octoprint (1).log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1712050/octoprint.1.log)

fdhosting, 2018-02-09:
By the way: maybe you can add in the topmenu of Octoprint next to the Temp/Humity display 
(Enc: 19.2°C Hum: 32.5%) also a fillament sensor display to see if for example in this setup the status of GPIO26 and GPIO19 ?

vitormhenrique, 2018-02-09:
Status is possible, i'm re writing the entire thing, I'll add on the next release.
I need access to your raspberry pi to trouble shoot this.. Shoot me an email... (email is on my github page)
I need you to open port 22 on your modem to remote ssh, and username and password...

fdhosting, 2018-02-09:
I have send you 2 emails ;-)

vitormhenrique, 2018-02-11:
The issue was caused because of the firmware. I don't know why, but octoprint could not send gcodes for your printer.

vitormhenrique, 2018-02-11:
I'm closing this issue, because it is not really a plugin bug, but an octoprint limitation.

closed 2018-02-11 by vitormhenrique