vitormhenrique/OctoPrint-Enclosure #93: Initial temperature upon octoprint start

koryba, 2018-02-09:
Hi,

I managed to add a temperature sensor and add a cooling fan at a particular GPIO (simple transistor switching circuit). The relation between the set temperature and the start of the cooling fan works really perfect - many thanks for that.

However, I have a question: Whenever the octoprint server starts I first need to manually set the temperature which I'd like to monitor in the "Enclosure Plugin"-tab. Is there a way to set a default value (e.g. 40 deg. of Celsius) upon startup without having the need of manual action?

In the config.yaml file I tried to use the "defaultTemp"- setting by changing it from "defaultTemp: 0" to "defaultTemp: 40". But unfortunately this didn't work.

Your help is much appreciated!

best regards
koryba

vitormhenrique, 2018-02-11:
It is a current limitation, it's going to be fixed on the next release.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and **implemented this requested** feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

koryba, 2018-03-01:
Dear Vitor,

I just tried it out. Fantastic. I love the possibility to add several temperature sensors as well as including DS18B20 sensors with different serial numbers.

However, at the moment I discovered two issues after a quick test
a) Once I save my inputs and outputs and make a restart of the Raspberry Pi, all inputs are gone and I need to re-configure them after startup.

b) Is there a way to have a temperature monitoring even when no print starts? 
What I mean is: once the input and outputs are created I need to manually set the target temperature in the temperature part of the "Enclosure Plugin"-tab.
If would be really great if it is possible that upon octoprint startup the target-temperature is taken from the "Default Value" of the temperature sensor. I mean this value where the text "Default temperature / humidity that temperature control will be set when the print starts." is written. In my case it is a cooler.

Thanks and best regards
Koryba

vitormhenrique, 2018-03-01:
I'm looking into (a) I think there is a better implementation from the one that I did. 

You will loose the settings again after I finish the release, but it should keep after that, I changed to much from the core settings, so migrating the data would be to much for me.

I'm implementing (b) now...  again, after you install it it will erase the current settings, but restarting the server should keep the last one...

vitormhenrique, 2018-03-01:
there it go, uninstall and install it again...

koryba, 2018-03-02:
Dear Vitor,

Fantastic. Many many thanks for this. All issues solved!

best regards
Koryba

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique