vitormhenrique/OctoPrint-Enclosure #94: Not working with more than 4 INPUT switches

derBerti, 2018-02-22:
Hi there!
I was filing through all the posts, but I could´t find a similar one which described my problem. Also I am quite new here and not a very good programmer... so please don´t raise your eyebrows when I acting like the absolute newbie who I am actually.
Now to the facts:
The line-up is OctoPi on a Raspberry Pi 3. 
Raspi, Printer-Board, Hotend, Heated and 2x LED strips are powered by a ATX power supply.
Also I installed a 8-channel relay-board for all this stuff and programmed everything (I am using 6 out of 8 relays) within the enclosure-plugin. All 8 OUTs are working perfectly over the Web-UI and I am able to happily switch everything ON and OFF. 
Also I connected 4 switches for ATX main power ON/OFF, printer ON/OFF and 2x LED strips ON/OFF to control everything at the server-housing.
Long story, but now comes the problem... 2 additional switches (one for filament detection and another one to restart the print after filament change) are simply not working at all. Everything was wired properly and the proper signals I can test by running "GPIO readall" in the command-line.
At the moment, these switches are programmed to toggle the remaining 2 relays only. Later they will have to run the printer commands like "pause" and "resume printing".
Meanwhile I was setting up OctoPi several times from the beginning, checked the wiring over and over and even wrote a small script to test the OUTs. It does´t work.

Is it possible, that there is only a limited amount of INs to be handled by the enclosure plugin?

Maybe the attached excerpts of the log file in combination with the "GPIO readall"-output and config.yaml helps to understand the problem?
[excerpt of log-file.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1749634/excerpt.of.log-file.txt)
[excerpt of config.yaml.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1749635/excerpt.of.config.yaml.txt)

Many thanks in advance!

vitormhenrique, 2018-02-22:
just to clarify, 

- all outputs work from the UI
- 4 "button" inputs work controlling GPIO
- 2 additional "inputs" don't work using filament detection of printer commands

Few questions:
Do the additional 2 inputs work when simply controlling other GPIO?
Do your printer runs marlin firmware of other firmware?

derBerti, 2018-02-23:
Thanks for your quick answer ...

- yes, all 8 OUTs work from the UI
- yes, 4 buttons work controlling the GPIO
- no, the other two buttons don‘t run printer commands (thats the plan for later), they also control GPIO (which doesn‘t work)

So we‘re only talking about controlling 6x Input GPIOs at the moment.

Anyway, the Firmware is Marlin on a Melzi-Board

vitormhenrique, 2018-02-23:
Is there any specific logic on the issue, for example, is it always the last 2 configured inputs that don't work?
if you remove the working 4 buttons, and just add the 2 that were not working, do they start working?

derBerti, 2018-02-23:
To make it easier (hopefully) you may see the setup while opening this issue and refer to the attached pic:
IN1 -> GPIO 17 (Pin 11), Switch S1, ATX ON/OFF for Relay K1
IN2 -> GPIO 27 (Pin 13), Switch S2, Printer ON/OFF for Relay K2 and Relay K5
IN3 -> GPIO 22 (Pin 15), Switch S3, LED strip 1 ON/OFF for Relay K3
IN4 -> GPIO 23 (Pin 16), Switch S4, LED strip 2 ON/OFF for Relay K4
IN5 -> GPIO 24 (Pin 18), Switch S5, Filament OK, Resume Print for Relay K7 testing
IN6 -> GPIO 25 (Pin 20), Switch S6, Filament detection for Relay K8 testing

OUT1 -> GPIO 05 (Pin 29), Relay K1, ATX ON/OFF (controlled by IN1)
OUT2 -> GPIO 06 (Pin 31), Relay K2, Printer ON/OFF (controlled by IN2)
OUT3 -> GPIO 13 (Pin 33), Relay K3, LED strip 1 (controlled by IN3)
OUT4 -> GPIO 19 (Pin 35), Relay K4, LED strip 2 (controlled by IN4)
OUT5 -> GPIO 26 (Pin 37), Relay K5, background lighting LS2 of switch S2 (controlled by IN2)
OUT6 -> GPIO 16 (Pin 36), Relay K6, -not yet programmed - background lighting LS5 of switch S5 (controlled by IN6)
OUT7 -> GPIO 20 (Pin 38), Relay K7, for testing of switch S5 (controlled by IN5)
OUT8 -> GPIO 21 (Pin 40), Relay K8, for testing of Switch S6 (controlled by IN6)

1.) 
As per your suggestion I removed the existing IN1, IN2, IN3, IN4 via Enclosure setup (by clicking the trash bin symbol). 
Now only the two "problematic" IN5 and IN6 were left over (without changing any parameter of these two). 
The setup is:
IN5 -> GPIO 24 (Pin 18), Switch S5, Filament OK, Resume Print for Relay K7 testing
IN6 -> GPIO 25 (Pin 20), Switch S6, Filament detection for Relay K8 testing
Now they are working!

2.) 
Next step is to add again (over the UI) the first two INs. The configuration is now:
IN5 -> GPIO 24 (Pin 18), Switch S5, Filament OK, Resume Print for Relay K7 testing
IN6 -> GPIO 25 (Pin 20), Switch S6, Filament detection for Relay K8 testing
IN1 -> GPIO 17 (Pin 11), Switch S1, ATX ON/OFF for Relay K1
IN2 -> GPIO 27 (Pin 13), Switch S2, Printer ON/OFF for Relay K2 (only)
That´s working also.

3.)
Now I am adding IN3 and also OUT5 controlled by IN2:
IN5 -> GPIO 24 (Pin 18), Switch S5, Filament OK, Resume Print for Relay K7 testing
IN6 -> GPIO 25 (Pin 20), Switch S6, Filament detection for Relay K8 testing
IN1 -> GPIO 17 (Pin 11), Switch S1, ATX ON/OFF for Relay K1
IN2 -> GPIO 27 (Pin 13), Switch S2, Printer ON/OFF for Relay K2 and Relay K5
IN3 -> GPIO 22 (Pin 15), Switch S3, LED strip 1 ON/OFF for Relay K3
This is working, too.

4.)
Next try…
IN5 -> GPIO 24 (Pin 18), Switch S5, Filament OK, Resume Print for Relay K7 testing
IN6 -> GPIO 25 (Pin 20), Switch S6, Filament detection for Relay K8 testing
IN1 -> GPIO 17 (Pin 11), Switch S1, ATX ON/OFF for Relay K1
IN2 -> GPIO 27 (Pin 13), Switch S2, Printer ON/OFF for Relay K2 and Relay K5
IN3 -> GPIO 22 (Pin 15), Switch S3, LED strip 1 ON/OFF for Relay K3
IN4 -> GPIO 23 (Pin 16), Switch S4, LED strip 2 ON/OFF for Relay K4

Here, the last added IN4 is NOT working! 
Tried with „GPIO readall“ while pressing the button and GPIO23 turns to LOW („0“) as it should.
Now, with this procedure I came „one switch“ closer than before, yay, but well… what can we try next?
[image]

vitormhenrique, 2018-02-23:
I must say that I have never tested that many inputs, I'll put together a test and update you.

I'm in the middle of completely refactoring the plugin, making it a bit more reliable and customizable. I've put more than 80 hours on it, but it's not quite ready yet. I've tested buttons and temperature control, but I t still need to finish the rest. 

I'll test on my new source base, if it works, I'll give you instructions on how to install, and let's see if that fix your issue.

derBerti, 2018-02-23:
All right then, no chance at the moment?
However, it‘s great to know, that you are reworking your plugin. 
IMHO its a top-notch supplement for Octoprint and for sure I am only one out of many who really appreciate your work!
Please let me know, if I can be of any support. And it‘ll be great to be one of the first to test it.

So I‘m goin to close this issue now. If you come across any idea that might help me until then, please contact me.

vitormhenrique, 2018-02-23:
Let's leave open so I don't loose track of this issue. I'll do some testing this weekend, I'll update if I have any thing to add...

derBerti, 2018-02-24:
All right, thanks! I‘m quite excited, what you might find out...

vitormhenrique, 2018-03-01:
I tried on the development brach and could get more than 7 inputs working,

Could you please try it out:

Uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

derBerti, 2018-03-03:
First off all... congrats! Your new version I really like. Unfortunately something went wrong while testing and I the file system broke. There was a Script of mine also accessing GPIOs. So I have to set up OctoPi again and I will test everything and give you some feedback!

vitormhenrique, 2018-03-07:
Let me know after you re-create your image if it is working for you.
I'm closing this issue for now, but you can re-open if needed

closed 2018-03-07 by vitormhenrique

derBerti, 2018-03-08:
No need to reopen! I just tested and was also wondering why I could not find the toggle function for GPIO Ins. But... this is settled as well, I just read. So, also from my side thanks a lot for the great work!