vitormhenrique/OctoPrint-Enclosure #95: [Request] DHT22 humid sensor triggers GPIO output

maukcc, 2018-02-26:
We are using the octoprint-Enclosure for our filament storage and drying unit.
We would like it when the humidity goes over xx%, our dehumidifier goes on, and when it goes under yy% the dehumidifier goes off.

vitormhenrique, 2018-03-01:
I just finished re-writing most of the plugin and implemented this requested feature, tested 95% of the plugin.

If you want to test it out uninstall the current plugin, go to the plugin manager, select get more, and where it says "from URL" use the development branch:

https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip

More documentation here:
https://github.com/vitormhenrique/OctoPrint-Enclosure/tree/dev

I'll push this to production after more people test it out.

vitormhenrique, 2018-03-07:
Implemented on V4.0

closed 2018-03-07 by vitormhenrique

vitormhenrique, 2018-04-12:
have you guys tested the functionality? everything working with the new version?

maukcc, 2018-04-13:
Not yet. we are doing mechanical re-designs 1st. and then electronics and software.
It is planned for end of May.

maukcc, 2018-06-22:
We are testing now. 
We are just wondering about "Value Deadband" 
I take it that this is the hysteresis on the setpoint, but it does not seem to work for the humidity.

so with a Value Deadband of 3 the behaviour should be:
When I set the humidity to 40% it goes down to 39.9% and shuts itself off. Then when it gets to 43% it goes on again. 
At this point it goes on again when it reaches 40%

maukcc, 2018-06-22:
Also: every time octoprint starts we need to set the target again.
We would like the option that the target stays

maukcc, 2018-06-22:
Or maybe a good idea to set fixed targets (that do not get nullified with a system restart) in settings, like
High target and Low target (to fix hysteresis issues)

vitormhenrique, 2018-06-22:
It's doing exactly what is supposed to do, instead of a strict control on the 40% it has a "dead band" from 43 to 40, where it will keep the last status..
read this: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/130
And there is a option startup with server to start the tempertature control when octoprint starts. Then it will set temperature control to the default value automatically.

maukcc, 2018-06-25:
I see what is happening. if we set the heater to 40 then it heats up to 40, cools down to 37 and starts heating again. This is good for the heater. but the dehumidifier works the other way around. it goes down to 40, stops working and goes up again, it will never get to 37 :) 
So for heater the start trigger is "set temp - deadband" . for dehumidifier it should be "set humidity + deadband"

maukcc, 2018-06-25:
I tried setting -3 in deadband, but that did not work :)

maukcc, 2018-06-25:
setting "temp_ctr_default_value" in config yaml helped getting the value for dehumidifier in at boot automatically.
Would be nice if you could set this in settings.

vitormhenrique, 2018-06-25:
but you should, it is the "default value" if it's not showing on the settings it might be a bug.

maukcc, 2018-06-26:
When I select Heater we can fill in deadband and maximum value.
With cooler and dehumidifier only deadband.
nothing else lists in settings

vitormhenrique, 2018-06-27:
default value was only being showing when "auto startup" is selected, it shows up on top of the dead band and the maximum value. I'll fix that issue...