vitormhenrique/OctoPrint-Enclosure #102: Does not work after update

modidep, 2018-03-07:
Hi I enjoyed using enclosure plugin but after ver 4.0 nothing works and cannot set it up. I want to revert back to 3.81 where can I find it.

vitormhenrique, 2018-03-07:
Can you send me your octoprint log?

vitormhenrique, 2018-03-07:
Also, please always follow the [Troubleshooting-Guide](https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/Troubleshooting-Guide) 
If you don't give me enough information I'll not be able to help.

vitormhenrique, 2018-03-07:
By the way, you need to refresh the page after the upgrade.

modidep, 2018-03-07:
Thanks for the prompt response,

I will send the log. I did a restart after installation. Then uninstalled and reinstalled the plugin. Thought maybe the problem is because of the browser tried with both Chrome and Firefox. Setup interface is messed up. I will send screenshots and the log in about 2 hrs.

Cuneyt

vitormhenrique, 2018-03-07:
on chrome, press control + shift + j, go to the console tab, and refresh the page, check what the console is spiting and send that as well.

modidep, 2018-03-07:
Unfortunately I have disabled the plugin and started a print. It will end in approx 2 hrs. I will send all of them then.
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1790535/octoprint.log)

vitormhenrique, 2018-03-07:
I does look like the plugin got upgraded, but you were still using an old "ui" screen, before the refresh.
Best way to fix this is to delete all settings from the plugin.
ssh on you pi open your configuration file

` nano .octoprint/config.yaml`

Go to the first line where you see **enclosure** and press `ctrl + shift + 6`, now keep going down (with arrow down button on keyboard) until you select all enclosure related configuration.

[image]

Now press `ctrl + k` to remove the lines, `ctrl + x` to exit, and `Y` to save.

Chose all browsers and reopen octoprint.

modidep, 2018-03-07:
Thanks a lot Vitor,

Will try and let you know. Thanks for your prompt help and support.

vitormhenrique, 2018-03-07:
don't forget to update with the results...

modidep, 2018-03-07:
Unfortunately after I delete the pertinent lines Octoprint does not turn on at all. I will look into it tomorrow.

vitormhenrique, 2018-03-07:
the log that you sent me had a bunch of erros from other plugins... you  can also send me the config file so I can see if anything else is missing....

another thing, you can even delete the file entirely, octoprint will recreate it for you.... (you will lose current configuration of course)

modidep, 2018-03-08:
Thanks Vitor,

I remembered I have a backup image from 2 days ago. What I can do is reflash the image and get copies of the octoprint.log and config.yaml before updating and then update properly by refreshing the browser and get copies of config.yaml and octopi.log afterwards if I have the same errors. I think that would work better. Is that correct?

From: Vitor de Miranda Henrique
Sent: 8 Mart 2018 Perşembe 02:15
To: vitormhenrique/OctoPrint-Enclosure
Cc: modidep; Author
Subject: Re: [vitormhenrique/OctoPrint-Enclosure] Does not work after update(#102)

the log that you sent me had a bunch of erros from other plugins... you can also send me the config file so I can see if anything else is missing....
another thing, you can even delete the file entirely, octoprint will recreate it for you.... (you will lose current configuration of course)

vitormhenrique, 2018-03-08:
Yes, that would be fine...

ashlintech, 2018-03-08:
Hi,  Your plugin is / was amazing, thank you.  Unfortunately having installed the update today it has erased all of my config for the plugin.  I'm hoping the update perhaps took a backup somewhere so that I don't have to rebuild it all from scratch.  I am mid build on my enclosure so have taken my own backup yet :-(

vitormhenrique, 2018-03-08:
Well, @ashlintech you did not read the release notes, it was **clearly** indicated that that would happen, and the update screen clearly indicate that you should always read the release notes. 
https://github.com/vitormhenrique/OctoPrint-Enclosure/releases/tag/4.00

ashlintech, 2018-03-08:
I did but unfortunately only for the Minor Release 4.01 - the rest didn't populate on the page until I have just clicked the releases button, so I didn't see the bit for release 4.00.

Oh well that will teach me to backup even a work in progress.

Thanks anyway - I still love the plugin :-)

vitormhenrique, 2018-03-09:
@modidep let me know if you fixed the issue....

modidep, 2018-03-09:
Thanks for your support Vitor,

I just flashed it basck with the image that has the version 3.81. I am planning to re-update (this time correctly) and update here this weekend.

From: Vitor de Miranda Henrique
Sent: 9 Mart 2018 Cuma 17:30
To: vitormhenrique/OctoPrint-Enclosure
Cc: modidep; Mention
Subject: Re: [vitormhenrique/OctoPrint-Enclosure] Does not work after update(#102)

@modidep let me know if you fixed the issue....

albebert, 2018-03-15:
updated, and nothing work,

when i try to add and ouput, i'm logged out from octoprint.

there is the log : 

2018-03-15 20:53:46,442 - tornado.access - WARNING - 404 GET /plugin/enclosure/getUpdateBtnStatus (127.0.0.1) 53.69ms
2018-03-15 20:53:47,089 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2018-03-15 20:54:23,120 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.200.100
2018-03-15 20:54:23,699 - tornado.access - WARNING - 401 GET /api/slicing (127.0.0.1) 61.13ms
2018-03-15 20:54:23,924 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.200.100
2018-03-15 20:54:28,407 - octoprint.plugins.multi_colors - INFO - on_api_command called: 'settings' / '{u'command': u'settings'}'
2018-03-15 20:54:31,157 - tornado.access - WARNING - 404 GET /plugin/enclosure/getUpdateBtnStatus (127.0.0.1) 50.25ms

vitormhenrique, 2018-03-15:
Got an email notifying of a new comment, but can't see anything new here.. weird...

modidep, 2018-03-15:
Sorry couldn't work on it yet. I was travelling.

Cuneyt Gursan

rhooper, 2018-03-17:
I wish there was an easy way to go back to the old version.
I can't get the plugin to work with my DS18B20 module any longer, since 4.0:
```

2018-03-17 02:36:02,917 - octoprint.plugins.enclosure - WARNING - An exception of type TypeError occurred on log_error. Arguments:
("'NoneType' object is not iterable",)

pi@octopi:~ $ cat /sys/devices/w1_bus_master1/28-0000069834ff/w1_slave
82 01 4b 46 7f ff 0e 10 70 : crc=70 YES
82 01 4b 46 7f ff 0e 10 70 t=24125

  enclosure:
    _config_version: 4
    rpi_inputs:
    - action_type: output_control
      controlled_io_set_value: low
      ds18b20_serial: 0000069834ff
      edge: fall
      filament_sensor_enabled: true
      filament_sensor_timeout: 120
      gpio_pin: '4'
      index_id: 1
      input_pull_resistor: input_pull_up
      input_type: temperature_sensor
      label: Enclosure
      printer_action: filament
      temp_sensor_address: ''
      temp_sensor_humidity: ''
      temp_sensor_navbar: true
      temp_sensor_temp: ''
      temp_sensor_type: 18b20
      use_fahrenheit: false
```

vitormhenrique, 2018-03-17:
You are missing the “28-“ on the serial...

rhooper, 2018-03-17:
Thanks @vitormhenrique.  PR sent with instruction updates!

modidep, 2018-03-18:
Hi Vitor,

We can close the issue. I still couldn't reinstall. I will open up anotger one if I can't upgrade it again. Thanks for your close attention to this and thanks again for your wonderful plugin.

Cuneyt

Reopened #102<https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/102>.

closed 2018-03-26 by vitormhenrique

modidep, 2018-04-07:
Hi Vitor,

Finally, I tried to upgrade again. I have uninstalled the plugin restarted everything and reinstalled it. Still no luck. Screenshots are attached. I tried different browsers, different operating systems, restarted server between each step and no luck. ;I am also attaching the log file as well.
[image]
[image]
[image]

vitormhenrique, 2018-04-09:
@modidep yeap, your screen shots are weird! lol...
Can you give me access to your octoprint? open a port on your router for both octoprint and your pi, send me an email (email is on my profile of github) 
I'll log on your system, fix whatever is going on, logout...

modidep, 2018-04-09:
Sure. We should arrange a time though. iw will email you.