vitormhenrique/OctoPrint-Enclosure #104: DHT 11 sensor frequently showing 0

StudioKamar, 2018-03-08:
Hello. I just updated to the latest version. when i configure everything which works perfectly before update i have some strange behaviour. My sensor shows correct temperature after a second it showing 0 after second again showing correct value and so on. Can You tell me where can be the problem ? For now because of that heater overheats the enclosure.

vitormhenrique, 2018-03-08:
I noticed that yesterday and already fixed on the master branch

closed 2018-03-08 by vitormhenrique

StudioKamar, 2018-03-08:
Thats great news because i knew that DHT11 has some problems but it work good till i updated. There is massive update lots of great functions really didnt want to downgrade :) There will be possibility to update automatically with that 4.01 version or i must do something ??

vitormhenrique, 2018-03-08:
it should appear on your octoprint instance, you can force check update as well.

vitormhenrique, 2018-03-08:
let me know if that fixes the issue for you...

StudioKamar, 2018-03-08:
I try tommorow because i have big print today and it ends just before midnight. I let You know

StudioKamar, 2018-03-09:
I force check the updates this morning and minor update apear. After update sensor readings are rock solid i check them later during the print but i think everything is good now. Thank You for Your assistance.