vitormhenrique/OctoPrint-Enclosure #107: Auto-Connect to printer.

DonGar, 2018-03-08:
Thank you very much for this plugin, and for the improvements over time!

Among other things, I'm using it to toggle power to the printer itself. It would be nice if the plugin could auto connect/disconnect the usb/serial connection to the printer after powering it up, and before shutting it down.

This is obviously a "nice to have" not a critical feature or a bug.

vitormhenrique, 2018-03-08:
I already implemented a "connect" button under input / printer / toggle. So you can add a button that when pressed will connect  octoprint to the printer.
It is not necessary to disconnect before shutdown, no harm for the printer.
And it is the scope of the plugin to connect to the printer when booting, this is a octoprint "thing"...
The plugin scope is to control inputs and outputs... not override default octoprint features.

DonGar, 2018-03-09:
If I understand the UI correctly, the "connect" behavior requires an external GPIO to be triggered to cause Octopi to connect.

I'm currently using your plugin to control power to the printer via the WebUI (and your Navbar support has made that much simpler to use). I have an output named "Printer" which controls power to the printer.

I don't see any way to use that input connect behavior to cause Octopi to connect without physical wiring, perhaps by adding a sensor for power supply output. Of course, it's just not that big a deal to have to hit the connect button after powering on the printer.

Also, I really do appreciate your work.

vitormhenrique, 2018-03-09:
yeap, you would need physical wiring, and that the scope of the plugin is to connect "outside" stuff to the octopi.

But i did find another plugin that can handle auto connect for you: https://plugins.octoprint.org/plugins/portlister/

vitormhenrique, 2018-03-09:
To use mine plugin, you would need to get a 3.3V output from your controller, and plug on a gpio on raspberry pi, when you turn it on, it would connect to the printer....

DonGar, 2018-03-09:
Oh... I could jumper from the GPIO out that controls the power supply to the input. That seems safer and easier.

If I were to get some time to work on it, and sent you a pull request to add something like this, would you consider it?

vitormhenrique, 2018-03-09:
But I don't quite get it, for me there are two scenarios:

1. You are using octoprint interface and using the enclosure tab turning the output on to turn on your printer. On this scenario, you only need to click connect that is on the same web interface, no need to even change tabs.

2. You are using buttons to control the enclosure, on this scenario you can already set up a button to connect the printer.

What am I missing? Can you explain a little bit more what is the use case?

DonGar, 2018-03-10:
It really isn't critical, it's just something that annoys me more than it should. The warning dialog from OctoPrint saying the printer disconnected unexpectedly at shutdown is much the same way.

It's annoying enough that I'm willing to implement the fix, but maybe not enough to learn how this plugin environment works in the first place.

closed 2018-03-11 by vitormhenrique