vitormhenrique/OctoPrint-Enclosure #109: [request] Observing Tool/Bed temperatures

EddyMI3d, 2018-03-08:
Hi!
Is it possible to get the tool/bed temperature as a value that can trigger an output?
Like the temperature sensor that can be connected to the GPIO.

Thanks in advance,
Eddy

vitormhenrique, 2018-03-08:
can you give me a example on what you want to do?
if you want to activate outputs based on your materials yuo can just use [g-code](https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/G-CODE-Control) commands.....

EddyMI3d, 2018-03-09:
E.g. for a hotend fan that runs only above a certain temperature of the hotend.
Or an indication (blink) light when the bed is cool enough to take off the printed part

vitormhenrique, 2018-03-09:
I don't quite get the use case yet, the extruder temperature is controlled by your printer, why would you turn on and off fans for it? For example all my lulzbot taz have fans on the extruder that need to be on 100% of the time.
For the bed temperature, it would not quite work, let's say you are starting a print, your bed temperature is at room temperature, so it would trigger the output already, instead of triggering when your print finish, better solution is at the end of the print, put a set and wait temperature on your print bed, and then send an enclosure gcode to turn a gpio on or off...

<pre><code>M190 R50 
ENC O1 S1
</code></pre>

EddyMI3d, 2018-03-09:
That's a neat hint. I think I'll stick with that. Thank you!

Akilestar, 2018-03-10:
I was actually looking for this too. Mainly to have a light on anytime the bed or extruder is above a certain temperature. If by setting the output type to temp control and then having the option to use the bed or extruder temp as the sensor along with gpio sensors.

vitormhenrique, 2018-03-11:
I'll leave this as a feature request now, but I spent a lot of time on the last release, more than 120 hours, and I need to move on to other projects, I'll only focus on bugs for the next few months. So don't expect new release with new features for quite some time.

R0dri, 2019-05-22:
any update? I found on the api reference that a simple get call to /api/printer would return actual temperatures of the bed and tools available. I don't think this to be to hard to implement right?