vitormhenrique/OctoPrint-Enclosure #111: Turn on and off USB Webcam

kodacy, 2018-03-09:
First! thank you for your wonderful job on OctoPrint-Enclosure plug-in Vitor.

One thin that would be great would be the capability of turning your USB webcam on and off by your plug-in.

On Octopi 0.13 it is quite simple with the SSH comands:
"sudo service webcamd stop" and "sudo service webcamd restart"

More info on: https://github.com/foosel/OctoPrint/issues/441

there is even ao coock book to do this on:
https://github.com/foosel/OctoPrint/wiki/Cookbook:-System-Commands

Any way to impertinent this commands directly in your plug-in?

vitormhenrique, 2018-03-09:
Definitely possible, but I spent a lot of time on the last release, more than 120 hours, and I need to move on to other projects, I'll only focus on bugs for the next few months. So don't expect new release with new features for quite some time.

kodacy, 2018-03-12:
Thank You! if i can help in any way just tell me.

vitormhenrique, 2018-06-20:
Implemented ability to run shell scripts from outputs.

closed 2018-06-20 by vitormhenrique