vitormhenrique/OctoPrint-Enclosure #113: filament sensor not working

dwc56, 2018-03-14:
i just updated to the newest version and all the configuration was erased sow i put it in so when a pin is low it will send m600 to the printer for filament change but it does not work if i change it to pause print that works but it will not send the gcode am i doing something wrong?

vitormhenrique, 2018-03-14:
Well, you are probably doing something wrong...

For the first issue you did not read the [release notes](https://github.com/vitormhenrique/OctoPrint-Enclosure/releases/tag/4.02) , it was **clearly** indicated that that would happen.
I spent more than 120 hours, of my free time, upgrading and coding this plugin with user requests, and would really appreciate if people would take 2 minutes to read the documentation...

Follow the [troubleshooting guide](https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/Troubleshooting-Guide), check the log, check what pin mode your raspberry pi is using, after all that, you should have opened a issue, and without print screens, log, it is basically impossible to help you...

dwc56, 2018-03-14:
do you need to do anything other then setting action to filament change if it works on pause printer

vitormhenrique, 2018-03-14:
do this: set it to filament, save, restart the server, start a "mock print" and press the filament sensor, cancel the print after few seconds, and send me the print screens and the log.

vitormhenrique, 2018-03-15:
any update? I tested the feature again yesterday and could get it working without issues....

dwc56, 2018-03-15:
[octoprint (2).log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1816415/octoprint.2.log)
[image]
[image]

vitormhenrique, 2018-03-15:
You are getting an error with the filamentreload plugin, did you configure the same IO on both plugins?
Disable filamentreload, try again, send me the log.

dwc56, 2018-03-15:
the pin number was set to -1 witch should have be disabled i uninstalled it and now i cannot change any settings and the save in spinin
[image]
g

dwc56, 2018-03-15:
i uninstalled and reinstalled and now it works