vitormhenrique/OctoPrint-Enclosure #115: Filament Sensor settings issues

jlg89, 2018-03-16:
I'm seeing two issues with Filament Sensor inputs on v4.04.

1. If I create two Filament Sensor inputs (on two separate IO pins, both tested functional per your troubleshooting instructions), and set both to trigger a Filament Change, it doesn't seem to happen unless both sensors trigger (or maybe just the second sensor, I haven't tested that permutation). My workaround is to make the second sensor a Pause action instead of a Filament Change.

2. On the Filament Sensor input, I set the detection timeout to 3600. But the plugin is sending two filament change gcode commands to the printer every minute. I also receive two IFTTT notifications per minute, for as long as the filament sensor is in a triggered state, so it's definitely coming from the plugin.

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1820602/octoprint.log)

[config.yaml.txt](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1820604/config.yaml.txt)

vitormhenrique, 2018-03-16:
I'm going out on vacation today and won't be able to test it out, I'll check when I'm back...

Also you are getting a lot of errors related with DHT temperature sensor, I highly recommend changing it to a DS18B20 (very inexpensive) or other supported sensor.
If not at least disable debug temperature logging, it will save some log space.

jlg89, 2018-03-16:
No problem, have a good vacation!

I didn’t have debug enabled, but apparently it’s enabled by default. I didn’t see any debug settings in config.yaml, so I enabled debug & temperature debug in the web settings, and then the plugin added these lines to config.yaml:

    debug: true
    debug_temperature_log: true

I went back in and changed them to “false”.

Note that I’m using “clean” plugin settings, because I had issues when upgrading to 4 (it didn’t fully wipe out the old config), and also to subsequent versions of 4, so I recently uninstalled the plugin, cleared all its settings from config.yaml, reinstalled & reconfigured.

vitormhenrique, 2018-03-17:
Awesome, thanks for letting me know, I'll get back to you when I have chance.

jlg89, 2018-03-19:
Same configuration files, I'm also having a problem with the Temperature/Humidity Control output starting automatically, even though the "start with server" option is un-checked. I have it set to auto-startup/shutdown with prints, but it starts with the server.

vitormhenrique, 2018-04-10:
@jlg89 Just a quick update, i have not had the chance to look into this, I got really busy with other projects and work, as soon as I have some time I'll test it out...

jlg89, 2018-05-10:
I changed the Filament Change timeout to 1800, and now it sends four commands per minute. So maybe there's something wrong in the math that converts the entered timeout value to milliseconds (or whatever time interval the backend is using).

vitormhenrique, 2018-05-11:
Reproduced the issue and fixed on https://github.com/vitormhenrique/OctoPrint-Enclosure/commit/34ae72a38051d9c9c9c88d59845325ac0096eb2c

You can uninstall and install the plugin again and it will have the fix for now, I'll push another release after I test couple more issues that people had sent...

closed 2018-05-11 by vitormhenrique

vitormhenrique, 2018-05-11:
also fixed on the new minor release.

jlg89, 2018-05-11:
Sweet. Thank you.