vitormhenrique/OctoPrint-Enclosure #117: Startup Delay no-longer working reliably after latest update

Leapo, 2018-03-18:
I have a "Regular IO" output, with "Startup Delay" set to 300 seconds, and it only seems to actually pull-down the output about 50% of the time. The previous version of the plugin worked every time, so this seems to be a bug introduced in 4.x

vitormhenrique, 2018-03-18:
You need to reproduce the issue and send a log, if not, it’s impossible to debug the issue.
Also, 
There are few things that can “cancel” the schedule io to trigger, for example cancel the print...

Leapo, 2018-03-18:
Steps to reproduce the issue:
1. Start a print.
2. Wait 300 seconds.
3. The GPIO state never changes.

Note: Manually clicking the button in the UI works, even after the timer fails to work, so the GPIO is configured correctly.

What type of log do you need? I'll dump it and upload it :)

jtcmedia, 2018-04-06:
I don't have the time delay, but I do have Auto Startup enabled and it has stopped working. Here is the pertinent part of the log:
```
2018-04-06 11:22:54,066 - octoprint.plugin - ERROR - Error while calling plugin enclosure
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/plugin/__init__.py", line 225, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1131, in on_event
    self.run_tasks()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure/__init__.py", line 1168, in run_tasks
    task['thread'].start()
  File "/usr/lib/python2.7/threading.py", line 739, in start
    raise RuntimeError("threads can only be started once")
RuntimeError: threads can only be started once
```

vitormhenrique, 2018-04-06:
@jtcmedia restart the server you should get it back working... and I need step by step how to reproduce...

jtcmedia, 2018-04-06:
@vitormhenrique yup, you were right, a restart got it back working, thanks. The issue first started showing up when I enabled 'Show button on navbar' and toggled it manually a few times.

vitormhenrique, 2018-04-10:
@Leapo could you do me a favor? I'm very busy on other projects and haven't had the time to look into this, however, I just added a bunch of more logs to try point it out what is going on...
If you could please, uninstall and install the plugin, you will keep all your configurations, but just in case, make a backup file...
After that, use the plugin as you would enabling the debug under the advanced part.
With the new logs I'll be able to understand what is going on, maybe do a quick fix without needing to setup my raspberry pi again (all my 4 raspberry pi's are busy running different tasks)

vitormhenrique, 2018-05-11:
@Leapo are you still experiencing this issue? 
I tried on 3 raspberry pis and could not reproduce, even a raspberry pi 1 worked fine with long delays...