vitormhenrique/OctoPrint-Enclosure #120: Outputs as PI shell commands?

clintonefisher, 2018-03-27:
Hi Vitor, thank you for the excellent plugin!

I have a set of LEDs hooked up to my pi that I have a command-line to control.   It occurred to me that you could use a command-line "output" from the enclosure plugin as a "glue" to many other possible features and functions.

Perhaps a simple ON = [enter command-line here] and OFF = [enter command-line here].   Have you considered adding such a feature?   If so, did you run into any roadblocks or reasons not to do this?

I understand you're very busy and may not have time for feature enhancements for a while.   While I'm not a python expert,  I might be able to follow your gcode output example and roll this in myself. 

Any thoughts would be appreciated, and thank you again!!

clintonefisher, 2018-03-27:
Hmm..I guess I could use your GCODE functionality combined with the GCODE system commands plugin...
https://plugins.octoprint.org/plugins/gcodesystemcommands/
I will try this first  :)

clintonefisher, 2018-03-27:
Aaaaand...  workaround achieved!   Your GCODE buttons and 
Gcode System Commands did the trick!  
Would be a cool add for the enclosure plugin though..   But who has the time?  :)
Thanks again for all your work Vitor!

vitormhenrique, 2018-04-06:
No problem, sorry for the delay, 

I'm planing to have some scripting for the plugin, but not shell commands.

closed 2018-04-06 by vitormhenrique