vitormhenrique/OctoPrint-Enclosure #121: how can i use use GPIO pin as input for power failure detection

amrithmmh, 2018-04-02:
i will be measuring voltage and give it as input to GPIO pin
if signal is low i need to pause the print and get the current print coordinates and save it to a file or something 
Is this possible with your plugin?

thanks for making this awesome plugin

vitormhenrique, 2018-04-06:
No you can not.
I'm planing to make something like this, but reading my UPS voltage.
If you loose power, you need to have an UPS to have enough time to save the status before raspberry pi get's shut down. This new plugin will not be inside this one, it will be a separate solution.

closed 2018-04-06 by vitormhenrique

amrithmmh, 2018-04-11:
but my raspberry pi will be running on smartphone power bank so it will have time to pause the print

vitormhenrique, 2018-04-11:
Have you tested your setup? i think raspberry pi might crash if you loose power from the ac and by only being connected to a power bank. 
And that scenario raspberry pi might be to slow to detect and save the current position.
The ideal implementation, is where both raspberry pi and the printer are one a power bank (UPS) where can control and measure the battery level. when the AC is off, the printer should save the current position and move away from the print. If you only save the location, the printer nozzle temperature being close to the print will make a mark and ruin the print anyway.

amrithmmh, 2018-04-13:
@vitormhenrique i have tested another plugin(modified from the filament sensor) and it pauses the print when the power goes off( havent actually tested the hardware i manually triggered it to pause needs the voltage sensor and adc for raspberry pi (not yet received) )
yes i think what you suggest is the better alternative because the nozzle will still be hot and will ruin the print