vitormhenrique/OctoPrint-Enclosure #122: Filament Jammed Sensor - Rotary Encoder, Hall effect, Optical,

Everlanders, 2018-04-04:
I have a CR10s that I got to print some large ABS parts and really love this plugin, thanks.

I am however having problems with the filament jamming for unknown reasons, it's not tangled, it's just not moving, the gear has chewed a hole in the filament halfway through the night and it's air-printing. Frustrating because it had been printing fine for 20 - 30 hours.  I have 11 failed prints like this and I have an idea that I think will be simple to implement.

There are many ways to sense filament movement, the easiest being a cheap hall effect sensor on a wheel or a IR sensor on a pinwheel - I intend to modify a flow meter for water once I finish typing this message, but they mostly all use pulse output.

It would be very helpful if there was an option to monitor a pin for these pulses and when they stop, pause the print just like the loss of filament detection. Just one option, How long to wait after last state change.

Thank you.

Everlanders, 2018-04-05:
Last night I built 2 variations of these sensors, one using IR and one using a modified flow meter.

https://photos.app.goo.gl/Oumt1j2d1gzG0ZQC2

The flow meter just has it's paddles clipped off and replaced with a rubber grommet that acts as a pulley, grabbing the filament on 3 sides. The IR is better because it has more pulses per revolution - the flow meter only has 2. 

The Arduino counts the pulses every 30 seconds and if less than 1, opens the relay which is wired to the stock filament sensor. This is less than ideal because you need to jumper the relay to get it started. Would be better if the plugin could watch for it only while printing.

vitormhenrique, 2018-04-06:
I feel like using an encoder and checking if it is sniping is the best choice. right now I don't have the time for anything really on this plugin, I'll put on the road map, but what you can defenetlly do, is create a sensor, with an arduino pro mini, or just a bare ATtiny and program the sensor logic on it, if it stop sensing it would set a pin high or low, and one input pin to enable or disable the sensor.
On the plugin side, you configure the output as a regular filament sensor, and on the enable / disable pin configure it as an output on the plugin and to start it when the print starts.

Everlanders, 2018-04-06:
Version 3:

https://photos.app.goo.gl/L3ogR4uD734MpQl22

I've got a small Arduino dealing with it, but I thought it would be a useful thing for others, more plug and play.

Thanks again!

vitormhenrique, 2018-04-06:
Are you using the ir sensor? have a link for the parts?

Everlanders, 2018-04-06:
You bet! [eBay Link](https://www.ebay.com/itm/5PCS-Slot-Type-IR-Optocoupler-Speed-Sensor-Module-LM393-for-Arduino/132478779559?hash=item1ed859c0a7:g:nQ0AAOSwScdZwOeI)

vitormhenrique, 2018-04-06:
How about the wheel? We would need to make a instruction on how to create the sensor, and document parts etc...
I have access to a laser cutter and can create a wheel, but not everyone has...

Everlanders, 2018-04-06:
I salvaged it from a Roomba, but it could easily be 3D printed. I can make a video if the plugin supports it.  ;-)

I'll put together some files tonight.

maukcc, 2018-06-21:
The MK4duo firmware has this option already installed (for up to 6 extruders) called Extruder Encoder Control.
We have not tested it, but they say it works.

jlg89, 2018-07-09:
There has to be a configurable delay before activating the sensor when the print starts, because the filament won't move during things like homing, bed leveling, etc. that typically occur at the beginning of a print. So we need an input with a "startup delay" like the outputs already have.

vitormhenrique, 2018-07-09:
Probably the best to implement is not a delay, because you would need to tweak the time to get it just right, I think the best implementation is "after the first change" on the encoder the filament sensor is activated, and a timeout of some minutes if it does change at all to catch you starting a print with a jammed extruder.
**BUT** don't have a nice setup to test and implement and I have been spending a lot of money on other projects that I'm working, so I don't plan to buy new encoders / sensors to implement this, if someone works a decent prototype for a filament sensor with encoders and ship it to me I'll gladly take the time to implement, but if not we will have to wait until I finish current projects....

jlg89, 2018-07-09:
I bought extra encoder sets. If I come up with something interesting, I'll send you one.

jlg89, 2018-07-13:
Here's a shell script that can be run on the Raspberry Pi, and triggered by an Enclosure Plugin action if desired. Working on a couple of different hardware solutions, one that tracks the filament itself (like a traditional switch-based filament sensor) and another that monitors spool movement. Will post details later. Both are Hall effect rigs, but any Pi-compatible pulsing output (optical, mechanical, etc.) could be used with this script.

<pre>
#!/bin/bash
#
# FilamentMonitor.sh -- Jon L. Gardner
#
# Monitors a Raspberry Pi GPIO input for pulses from a filament motion sensor.
# Triggers a print pause via OctoPrint API, and optionally notifies via IFTTT.
# Requires inotify-tools (in OctoPi, sudo apt-get install inotify-tools).

# set timeout in seconds
timeout=10

# select input pin to monitor for motion sensing
GPI=5

# OctoPrint API key
# (only required for pause option 1, see below)
apikey=YOUR_OCTOPRINT_API_KEY

# IFTTT API key and event name (from Enclosure Plugin settings)
ifttt_key=YOUR_IFTTT_API_KEY
ifttt_event=YOUR_IFTTT_EVENT_NAME

# prepare the pin
if [[ $GPI ]]; then
 /bin/echo "Configuring input on pin ${GPI}"
 if [ ! -d /sys/class/gpio/gpio${GPI} ]; then
  /bin/echo "${GPI}" > /sys/class/gpio/export
 fi
 /bin/echo "in" > /sys/class/gpio/gpio${GPI}/direction
fi

function trap_ctrlc ()
{
  /usr/bin/printf "\n\nLongest pulse time was $maxtime seconds.\n\n"
  exit 2
}

trap "trap_ctrlc" 2
init=1
maxtime=0

while :
do

 if [ $init == 1 ]; then
  init=0
  # wait for the first change before starting the timer
  /bin/echo "Waiting for initial filament movement..."
  /usr/bin/inotifywait --quiet --event modify /sys/class/gpio/gpio${GPI}/value >/dev/null
 fi

 # once things are moving, detect if things stop moving for longer than the timeout
 stime=`/bin/date +%s`
 /bin/echo "Motion detected, timer reset. Waiting for a max of $timeout seconds..."
 /usr/bin/inotifywait --timeout $timeout --quiet --event modify /sys/class/gpio/gpio${GPI}/value >/dev/null

 if [ $? == 2 ]; then
  /bin/echo "Filament has stopped moving. Pausing print..."

  # pause the print via OctoPrint API (assumes localhost)
  /bin/echo '{"command": "pause" , "action": "pause"}' | /usr/bin/curl -X POST -d @- -H "X-Api-Key: "$apikey"" -H "Content-Type: application/json" http://localhost/api/job

  # notify via IFTTT
  if [[ $ifttt_key ]]; then
   /bin/echo '{"value1": "Printer paused by filament motion sensor." , "value2": "Longest pulse delay thus far is $maxtime seconds."}' | /usr/bin/curl -X POST -d @- -H "Content-Type: application/json" https://maker.ifttt.com/trigger/$ifttt_event/with/key/$ifttt_key
  fi

  # wait for the timeout period, then start over
  /bin/sleep $timeout
  /bin/echo "Resetting status..."
  init=1

 else
  #  maintain record of longest pulse time
  ftime=`/bin/date +%s`
  ttime=`/usr/bin/expr $ftime - $stime`
  if [ $ttime -gt $maxtime ]; then
   maxtime=$ttime
  fi

 fi

done
</pre>

vitormhenrique, 2018-07-13:
@jlg89 thanks for the script, despite is helpful as a general guide on how to implement, if you can convert it to python it would be much more helpful, because it would be easier to port.

Also, python has some nice features for the GPIO, like interrupts for the pins, making easy to get exact time that the GPIO change status.

jlg89, 2018-07-25:
@vitormhenrique if you can give me a shipping address, I'll send you one of the pulse-generating filament sensors I built.

jlg89, 2018-07-26:
[Filament Motion Sensors.pdf](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2233797/Filament.Motion.Sensors.pdf)

vitormhenrique, 2018-07-31:
@jlg89 send me an email on the email linked to my profile and I'll give you my address....

starbuck93, 2018-09-03:
@jlg89 Thanks for that writeup, you should make a new repo for it! I just bought a few of the https://www.pololu.com/product/3081 sets to try to use your enclosure with a Duet board.

jlg89, 2019-03-26:
OK...finally, here's my first shot at a Python version of the script. This is incomplete, missing the printer pause & notification actions, but the basic pulse input detection is there.

`
#!/usr/bin/env python2.7

import sys
import time
import urllib2
import json
from collections import namedtuple
import RPi.GPIO as GPIO

GPIO.setmode(GPIO.BCM)

CHAN=5
TIMEOUT=30
OCAPIKEY='your_OctoPrint_API_key'

GPIO.setup(CHAN, GPIO.IN, pull_up_down=GPIO.PUD_UP)

global MOV
MOV=0
RUN=0
DONE=0

def my_callback(channel):
    global MOV
    MOV += 1

GPIO.add_event_detect(CHAN, GPIO.BOTH, callback=my_callback, bouncetime=300)

def PrintDone(apikey):
   # check OctoPrint API for main tool set temp, if 0.0 assume job finished
   url="http://localhost/api/printer/tool?apikey={0}".format(apikey)
   response = urllib2.urlopen(url)
   html = response.read()
   x =json.loads(html, object_hook=lambda d: namedtuple('X', d.keys())(*d.values()))
   tempt=x.tool0.target
   if tempt==0.0:
    return 1
   else:
    return 0

try:

   while DONE==0:
    MOV=0
    RUN=0
    print('Waiting for initial pulse.')
    while RUN==MOV:
     # wait for initial pulse
     time.sleep(.1)
    print('Pulse detected.')
    # monitor I/O for state changes
    while True:
     start_time = time.time()
     MOV=0
     RUN=0
     while RUN==MOV:
      if time.time() - start_time > TIMEOUT:
       if PrintDone(OCAPIKEY):
        print('Print appears to be finished.')
        DONE=1
        break
       # timeout actions go here -- pause print, notify via IFTTT etc.
       break
     # output a dot at each state change
     sys.stdout.write('.')
     sys.stdout.flush()

except KeyboardInterrupt:
    GPIO.cleanup()       # clean up GPIO on CTRL+C exit
GPIO.cleanup()           # clean up GPIO on normal exit
`

sciaio, 2019-07-12:
Really nice, are there any more progress? Will be available into the plugin soon?

peterloron, 2020-04-06:
Hi. I'm going to play with the script @jlg89 posted (thanks!), but wondering what the status of this functionality is in the main plugin? Thanks!

vitormhenrique, 2020-04-06:
The status is really really low priority, I started masters on computer science and have been busy with other projects, I plan to re-write the entire base Conde of the plugin to make it easier to understand and expand, so maybe more people would be able to push commits. 
I wanna re-write the code with a better objected oriented mentality and add some of the features that are more commonly requested. Unfortunately it's not the case of this. One day I might get to this and figure a way to implemented, but honestly is not on a close road map.