vitormhenrique/OctoPrint-Enclosure #124: AM2302 Temperature/Humidity Readings not showing up on Octoprint HUD

logordon, 2018-04-08:
Hi vitormhenrique,

I Just installed your "Enclosure Plugin" via your instructions. I have the AM2302 running on a Raspberrypi 1. I followed your instructions and the temperature and humidity sensor is not reading to Octoprint. On terminal running "sudo ./AdafruitDHT.py 2302 4"  spits out the temperature and humidity sensor readings so I can confirm that the sensor is working. I followed the instructions below:
***
You need to install Adafruit library to use the temperature sensor on raspberry pi.

Open raspberry pi terminal and type:

cd ~
git clone https://github.com/adafruit/Adafruit_Python_DHT.git
cd Adafruit_Python_DHT
sudo apt-get update
sudo apt-get install build-essential python-dev python-openssl
sudo python setup.py install
You can test the library by using:

cd examples
sudo ./AdafruitDHT.py 2302 4
Note that the first argument is the temperature sensor (11, 22, or 2302), and the second argument is the GPIO that the sensor was connected.
***

[image]
[image]

logordon, 2018-04-08:
Not sure what happened, but it works! Thanks!

closed 2018-04-08 by logordon

vitormhenrique, 2018-04-09:
Yeap, as I mention on the read-me DHT sensors and raspberry pi don't work very well together, sometimes the sensor will fail to read for minutes! 
I strongly suggest getting a very cheap DS18B20 one wire temperature sensors, you can have multiple on the same wire bus and they are very reliable.