vitormhenrique/OctoPrint-Enclosure #126: [Feature Request] RGBW LED support

Zingabopp, 2018-04-17:
As the title says. I have a strip of SK6812 LEDs and for each pixel they have 4 LEDs, one of which is white. Currently, I can support it by having the Arduino set the strip to white if it receives a certain RGB value, but it would be nice of OctoPrint-Enclosure supported it directly. It looks easy enough to add to neopixel_indirect.py, but I wouldn't know how to go about adding it to the web interface.

vitormhenrique, 2018-04-17:
SK6812 should be a drop in replacement to WS2812B (that is already supported), setting the strip to white should be as simple as sending a rgb(255,255,255) from from the user interface needs to be updated.
The only thing that needs to be updated (you can even connect the strip directly on raspberry pi) is changing [this](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/neopixel_direct.py#L24) or [this](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/extras/octoprint_neopixel/octoprint_neopixel.ino#L93) line of code, depending if you want to connect the strip directly or indirectly, to get a RGBW color like:

` Color(0, 0, 0, 255)`

You can do this for now and get it working from the UI, on the future I can add the SK6812 as settings and implement the changes to handle it, but now I'll not add new features on the plugin. 
Last release was pretty big with more than 2 months of development and focusing on this software, now I'm working on new projects and will only handle issues on this plugin.

shbatm, 2018-09-20:
I would like to tag onto this (and as an extension of #141): it would be nice to be able to control an SMD5050 12V RGB strip with this plugin instead of just NeoPixels.  I have an external controller board that I use to drive the 12V strips with 3 pins from the Pi (via some N-Channel MOSFETS), but I would like to take advantage of the GCODE & color options, rather than just add 3 PWM outputs.  I was hoping to blend in something like what you already have with the NeoPixel options and the [LEDStripControl](https://github.com/google/OctoPrint-LEDStripControl) plugin.  The two don't play well with each other.

If I get a chance over the next few months, I may attempt to make the integration myself and send you a PR... just wanted to leave my intent here in case someone is already attempting this or I've missed the option somewhere in the settings.

rosskinard, 2018-10-31:
I'd like to add to the request for basic SMD5050 RGBW strip support vs NeoPixels. I plan to put magnetic sensors on the door(s) of the enclosure so it would be nice to have the plugin read the change in the door sensor and turn on bright-white lighting while open, but have any level of color or white LED controls based on job status, temp, etc. I'm using a small, external controller like @shbatm is to let the low-voltage signals from the Pi drive the 12V strip.

Mavi222, 2019-03-03:
I would also like RGBW support. I have RGBW LED strip, connected to my Raspberry Pi through TIP 122 transistors, which means that I have each color on different GPIO pin. It would be awesome to implement and option to set RGBW with individual pins, and to be able to choose the color with the color wheel that is available with the RGB Led Strip option (which sadly uses only one pin)

vitormhenrique, 2019-03-04:
@Mavi222 I don't have plans to implement those old strips supports with the color wheel. You can still use the plugin to control the color, setting 3 outputs (one for each gpio of the strip) and controlling he colors individually, you can also create "shortcut buttons" with mixed colors, like (orange, white, etc) using gcode outputs.
My decision can change on the future, but for now, there are other things that are my priority for the plugin. Also, anyone can fork the plugin, implement the feature and I'll accept push requests.