vitormhenrique/OctoPrint-Enclosure #127: Two DHT22 sensors do not display well

stusona, 2018-04-17:
I have one DHT22 sensor called "ambient, and one called "filament", and they do not display as expected.

[image]

Great plugin! Thanks for all your work!

stusona, 2018-04-17:
You betcha. I figured the logs were irrelevant, but I guess I should have just included them.

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1920501/octoprint.log)

[image]

[image]

vitormhenrique, 2018-04-17:
That is so weird,just as a test, if you delete both temperature sensors, and two dummy SI7021 sensors, what does it do?

stusona, 2018-04-17:
I deleted both and created two SI7021 sensors without addresses:

[image]

vitormhenrique, 2018-04-17:
Thanks, I'll try it out tonight, I tried to access my development raspberry pi from work but they blocked remote ssh connections...  
I'll update when I test it out...
Also noticed that you are using another skin for octoprint, can you tell me what is it? so I can see if it is related or not (probably not)

stusona, 2018-04-17:
Thanks. I'd help but I don't know anything about javascript, css, etc. Maybe I should take an online course or something.

I'm using [Themeify](https://github.com/Birkbjo/OctoPrint-Themeify).

vitormhenrique, 2018-04-18:
just fixed on master branch, just uninstall and install the plugin again, all settings should be ok...

closed 2018-04-18 by vitormhenrique

stusona, 2018-04-19:
Thanks Vitor!