vitormhenrique/OctoPrint-Enclosure #129: Table for temperature sensors is not correctly formatted

dogbert1974, 2018-04-24:
Hi,

I've added 5 temperature sensors (DHT22) and found that the table formatting has some visibility error.
All the sensor are on the same line making the data unreadable:
[image]
Maybe there are only some tags missing in the table.

p.s. should be nice to have a graph on the same page showing a graph of the data during time.

vitormhenrique, 2018-04-24:
Hello, I just fixed this formatting issue few days ago, but I did not put a new release, so you  uninstall and install the plugin again, all your settings will remain.

For the temperature charing, that feature was already requested:

https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/34

And I'm waiting the issue https://github.com/foosel/OctoPrint/issues/321 on octoprint itself be implemented so I can add temperature sensors to octoprint chart.

dogbert1974, 2018-04-24:
Fantastic. Just checked online and found the update available. Now it works fine.

I really like this plugin.

Thank you very much.

vitormhenrique, 2018-04-24:
No problem!
Thanks!
Just a heads up, having 5 dht sensors will probably make your pi work hard to get all the values.

I really encourage you getting other temperature sensors, if you only need temperature DS18B20 is chainable with one wire communication.

closed 2018-04-24 by vitormhenrique

dogbert1974, 2018-04-24:
I'm really going to use only three of them. 5 was to check the gpio settings. I will later add a power switch