vitormhenrique/OctoPrint-Enclosure #131: Failing sudo is not detected

yschaeff, 2018-04-27:
It took me a while to discover why enclosure would not read my DHT11. 
It logged:

2018-04-27 22:27:36,836 - octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)
2018-04-27 22:27:36,838 - octoprint.plugins.enclosure - INFO - Sensor Temp Temperature: 0.0 humidity 0.0

Ultimately it appeared enclosure tried to run the command with sudo. While the octopi user is allowed to read the sensor it is not allowed to sudo on my system. The error message was a red herring as it made me investigate the adafruit library instead. (So if the exception could indicate line and file that would be great)

def read_dht_temp() calles the temp script without checking the exit code. If it did it could print a meaningful log line.

yschaeff, 2018-05-02:
Sounds good.
Thank you for sharing your work. It is a great plugin!

vitormhenrique, 2018-05-11:
fixed on https://github.com/vitormhenrique/OctoPrint-Enclosure/commit/c72c5bf1f2b9ea0e40392fa2603118ade4a81fd0

closed 2018-05-11 by vitormhenrique