vitormhenrique/OctoPrint-Enclosure #132: [Feature Request] - Latching Switch Support

RitchRock, 2018-04-28:
It seems that only momentary switches are supported by button input. I would like to use a latching switch to, say turn a power relay on/off or lights on/off.

Currently, it seems to be only possible to trigger on a rising or falling edge from a momentary switch, however it would be nice to have the option trigger on both rising and falling edge when in "toggle" mode. So when the switch is pushed in, the action is toggled on, say a rising edge. Later, when the switch is moved back to the original position, the action is toggled on the falling edge. Basically, an option to toggle both on rising and falling edge.

vitormhenrique, 2018-04-28:
Not really, at least none without changing the code, I had problems when I used other event attachments other than rising and falling, that is why I only implemented like this, my suggestion is to just use a simple push button right now, I'll have to do more research and figure it out a good solution for a latching button.

Right now I'm not going to make changes on this plugin, I'm focusing on other projects, and only fixing issues, but I'll research and try to put it out on the next feature release.

RitchRock, 2018-04-28:
I completely understand. Thanks so much for your time on this.

vitormhenrique, 2018-06-14:
Added on the [new release](https://github.com/vitormhenrique/OctoPrint-Enclosure/releases/tag/4.09) update and test it out...
You should be able now to add two "edges" for the same button (you need to configure two inputs with the same GPIO)

closed 2018-06-14 by vitormhenrique

shbatm, 2018-09-18:
@RitchRock -- Is this working for you? I'm having the same issue and I tried adding two Inputs with the same GPIO pin and resistor selection, but opposite edges.  Only the 2nd edge added works (the one furthest down on the config page).

RitchRock, 2018-09-18:
I built my enclosure prior to the jun 14th update and haven't had time to add latching buttons, as it's working fine without them. The Enclosure Plugin, as whole, has worked great.