vitormhenrique/OctoPrint-Enclosure #133: TMP102 returns 0 in OctoPi with a working I2C and TMP102

andreiavram, 2018-04-28:
I'm using OctoPi 1.3.8 and Enclosure Plugin 4.06, and I can't seem to get values to be read in the interface. All the connections are alright, but I always get a 0 displayed - the results of an exception, as I can see in the code.

I've tried to run the tmp102.py file in the package, with the octopi virtualenv enabled, got an error that it couldn't find the smbus package. Tried to pip install smbus, got a response that there's no such package.

After some googling, I pip installed smbus2, and manually updated the reference to smbus2 in the relevant places of the tmp102.py file in the package, and now manually running python tmp102.py yields correct values.

Even so, in the interface I get a 0 reading. Not sure how to debug further ...

vitormhenrique, 2018-04-28:
Are you using raspberry pi? 
Also, can you copy both the output of your test on the terminal with the results, and also octoprint log?
Without logs and print screens is really difficult to help...

vitormhenrique, 2018-04-28:
Also, can you check what python version you are using?

andreiavram, 2018-04-28:
Using Raspberry Pi 3, Python 2.7.9. I couldn't find anything relevant in the octoprint log about the TMP102 - I have a switch set up with the enclosure plugin that works and that I see in the log.

OctoPrint version : 1.3.8
OctoPi version    : 0.14.0

I added a custom DEBUG level log in the octoprint logging settings page, and restarted both octoprint and the RPi, but I can't seem to get any output out of that. Is there something else i should set up for logging to work?

These are the only lines with "enclosure" in them:

`|  Enclosure Plugin (4.06) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_enclosure
2018-04-27 18:28:55,979 - octoprint.plugins.enclosure - INFO - Setting GPIO mode to BCM
2018-04-27 18:28:55,983 - octoprint.plugins.enclosure - INFO - Setting GPIO pin 14 as OUTPUT with initial value: 0`

[image]

Output on the pi:
`(oprint) pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ python tmp102.py
23.4375`

I checked that I'm using the correct vitrtualenv and it seems that I do:
`(oprint)pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ ps ax | grep octo
  456 ?        Ss     0:05 avahi-daemon: running [octopi.local]
  554 ?        Sl    24:43 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_raspicam.so -fps 10
 1618 ?        Sl    20:43 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000
`
Any idea on how to enable logging?

vitormhenrique, 2018-04-28:
There is one option inside of the advanced area of my plugin to enable debug information, but you should not need to change anything on octoprint itself. The log on the main page is not that reliable, I usually get the file from the default location, of stop the service and stat it manually so I can see in real time all messages from octoprint.

You can stop octoprint with "sudo service octoprint stop"

And start it without  the service, to be able to see the log in real time with "./oprint/bin/octoprint serve"

Question, you installed the smbus on the virtual environment? At runtime, the plugin execute python scripts on the main environment, not the virtual.

It's really weird that you can't find the smbus package, this is the output of my raspberry pi's:

```
pi@octopi:~ $ pip install smbus
Requirement already satisfied (use --upgrade to upgrade): smbus in /usr/lib/python2.7/dist-packages
Cleaning up...
```

Some people had problems when using really old images with his plugin, what is the version of your kernel? If it's a little older, I would suggest getting a clean octopi image and installing everything again (not hat huge deal usually, 30 minutes and you should be done)

```
pi@octopi:~ $ uname -a
Linux octopi 4.9.28+ #998 Mon May 15 16:50:35 BST 2017 armv6l GNU/Linux
```

vitormhenrique, 2018-04-28:
I just found that you can install smbus also from apt-get:

`sudo apt-get install python-smbus`

Also install i2c tools:

`sudo apt-get install i2c-tools`

And I also noticed that I forgot to porperly format the temperature on the tmp102.py, change line [45](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/ab97240ca2dc5fe96a9161ac9b07e8b285e668c5/octoprint_enclosure/tmp102.py#L45) to:

`print('{0:0.1f}'.format(temp))`

vitormhenrique, 2018-04-29:
Nice! Just fixed the read-me!

closed 2018-04-29 by vitormhenrique