vitormhenrique/OctoPrint-Enclosure #136: Enclosure state remembered through reboots

zbrozek, 2018-05-08:
Despite auto-startup being unchecked, I'm seeing that my heaters and fans are defaulting to the last setting upon Octoprint shutdown. This seems like a bug?

vitormhenrique, 2018-05-09:
It's by design, it will remember the last status, and if you reboot pi it will default to that.

Feature was added because of this: https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/27

zbrozek, 2018-05-09:
Could you please add an option to turn that feature off? It's the opposite of what I'd like to have happen, and has already caused me to have my enclosure unexpectedly hot for a week. I would much prefer the octoprint-enclosure to be as stateless as possible.

vitormhenrique, 2018-05-09:
But you have numerous ways to prevent that, you can have a auto shutdown after the print finishes (is is suggested, with bold warnings on the plugin tab), you can just turn off your heater, now you can even use g-code commands on your print to turn on and off the heater.
Remembering status is very good for lights, door locks, etc...
Either way I'm not changing or adding features for the plugin on a near feature, last release took me 2 months of coding and I'm working on new projects now, I'll consider for next release, but I it will not be on my top priority.

zbrozek, 2018-05-09:
I do have auto-shutdown turned on, but it assumes that enclosure state is solely connected to print status. The case that this bit was as follows:

- Some hand-written gcode wasn't right, and I punched the e-stop, killing power to everything.
- I turned off the printer switch.
- I reset the e-stop (so that my Octoprint server starts back up).
- I walked away before the server had finished booting.

A week later, much to my surprise, I find my enclosure has been sitting at 50 Celsius. No print had completed, and the (now undesirable) state had been restored in my absence.

vitormhenrique, 2018-05-09:
Got it, I think it's an overkill to have your raspberry pi shutting down like that, make your e-stop send a cancel signal to octoprint, it might still print for a few seconds, and if you really want to shut something down, shut down the printer.
I have corrupted many sd cards by not properly shutting down my raspberry pi.

vitormhenrique, 2018-06-14:
I did some testing, the plugin does not remember the last state of the gpios, unless you select the item to auto start with the server. I could not reproduce your issue.

vitormhenrique, 2018-06-14:
please test it out again and let me know, re-open the issue if necessary...

zbrozek, 2018-07-03:
Something similar just happened:

* No printer connected. Fans and heaters were off, though I had been turning them on and off to test them after replacing a fan.
* I rebooted from the console (ssh).
* Upon Octoprint restart, the Enclosure plugin restarted the heaters and fans, but targeting a temperature I hadn't used in several weeks. (!!)

Where is it even storing that state?

vitormhenrique, 2018-07-03:
You need to send me printscrens of your configuration and the log files, without them it is impossible for me to figure it out what is going on.

zbrozek, 2018-07-03:
Are these the right files?
[log_and_config.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2160042/log_and_config.zip)

vitormhenrique, 2018-07-03:
Take print screens from your configuration, I want to double check the your settings...

vitormhenrique, 2018-07-03:
what I would do is to delete and re-create the heaters outputs, that will purge and probably fix any data issue on your configuration.

zbrozek, 2018-07-04:
OK; here are screenshots before I go do that.

[image]
[image]
[image]
[image]
[image]

vitormhenrique, 2018-07-04:
ok, delete all settings and re-enter them, let me know if it fixes the error.

zbrozek, 2018-07-08:
I don't think nuking all of the outputs and inputs really nuked everything. I just had the same behavior again after rebooting Octoprint for the update to rc2. Here's what the Enclosure plugin status page looks like immediately after reboot:
[image]

vitormhenrique, 2018-07-08:
Try this: go to the settings, on the cooler and heater enable auto startup, it is going to show a "default value" field,  set that field to zero, save the settings, then you can disable the auto startup again.

vitormhenrique, 2018-07-08:
your system is doing something weird, the plugin set's those value as 0 when you create new outputs, somehow your's is setting to 60 and 58.
https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/static/js/enclosure.js#L388
my suggesting if this don't fix the issue is to re-install octoprint.

closed 2018-07-08 by vitormhenrique

zbrozek, 2018-12-23:
This continues to be a problem with a completely clean installation of Octoprint and the plugin and a from-scratch reconfiguration. On first boot of the system one of the heaters will reset to some historical value that is not what it was at the time of shutdown. With some new hardware that did not yet have diodes to force fans on when a heater was running, it melted a shroud. And there's also an issue where the web UI does not necessarily update to represent the status of the outputs faithfully.

In general this is pretty scary misbehavior and probably ought to get properly root caused. I'd love to help do that. What are next steps?