vitormhenrique/OctoPrint-Enclosure #137: Temp alarm: controlled IO field is not editable

bumbalu, 2018-05-09:
First of all thank you for Plugin!
When setting up temperature alarm output the "controlled IO" field is a drop-down field which has no pin numbers available in the drop-down list and does not allow to input any number. 
I use the 4.6 version of plugin. 
Thank you, Vitor.

vitormhenrique, 2018-05-09:
You need to configure a output fist, set it up, than it will show up on the drop-down.
For example, you set up a light emergency output, and configure it, give it a gpio number, etc etc etc..
Than you set up a temperature alarm, and you should be able to select what "controlled IO" you want...

closed 2018-05-09 by vitormhenrique

bumbalu, 2018-05-14:
Vitor, thank you for explanation! Now everything works fine.