vitormhenrique/OctoPrint-Enclosure #139: BME280 Temperature sensor does not work using latest Octopi Image 0.15.1

workshopninja, 2018-05-11:
I have a functioning BME280 sensor that is giving me the following error in the logs:

octoprint.plugins.enclosure - WARNING - An exception of type ValueError occurred on log_error. Arguments:
('need more than 1 value to unpack',)

I installed the packages using:

vitormhenrique, 2018-05-11:
try this and let me know the result, ssh to your pi, and go to the packages content:
`cd oprint/lib/python2.7/site-packages/octoprint_enclosure/`

Execute:
`python BME280.py 76`

What is the result?

Btw: don't activate the octoprint python env, everything needs to be on system python, not octoprint python.

workshopninja, 2018-05-11:
Wow - instant response!

So that works

workshopninja, 2018-05-11:
sorry 

pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ python BME280.py 76
16.1 | 55.7
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ 

so that works

vitormhenrique, 2018-05-11:
what about if you put sudo first?  `sudo python BME280.py 76`

workshopninja, 2018-05-11:
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python BME280.py 76
20.4 | 45.4
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $

workshopninja, 2018-05-11:
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ sudo python BME280.py 76
20.4 | 45.4
pi@octopi:~/oprint/lib/python2.7/site-packages/octoprint_enclosure $ 

with sudo

workshopninja, 2018-05-11:
The Output Relays and the Input Filament sensor DO work.

vitormhenrique, 2018-05-11:
Can you send me an email with your ip, ssh info (username and password) and open the ports for me to look at?
If the command line is spitting the value for the script I have no idea why the plugin is not getting it...

Let's try one more thing before I log in... enable both "debug" and "temperature log" on the advanced sections of the plugin. Delete current log, restart the server... it will log the command that octoprint is executing to get the value, let's see what is happening...

workshopninja, 2018-05-11:
Here is the new log file:
[octoprint-4.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1996666/octoprint-4.log)

vitormhenrique, 2018-05-11:
I think I figure it out, let's see, the command that octoprint is sending is 
`sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/BME280.py 0x76`

Note that it is sending "0x76" it should send 76. Can you go to your settings and change the address to just "76"?

workshopninja, 2018-05-11:
After posting I noticed that - changed it and same result in browser

vitormhenrique, 2018-05-11:
how about the log?

workshopninja, 2018-05-11:
[octoprint-5.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/1996673/octoprint-5.log)

vitormhenrique, 2018-05-11:
and if you just run manually the command on ssh terminal? 
`sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/BME280.py 76`

workshopninja, 2018-05-11:
pi@octopi:~ $ sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/BME280.py 76
19.6 | 48.3
pi@octopi:~ $

vitormhenrique, 2018-05-11:
yeah, I need the ssh stuff....

workshopninja, 2018-05-11:
OK - don't know how to do that...

vitormhenrique, 2018-05-11:
don't worry, send me an email, I'll help you out...

workshopninja, 2018-05-11:
Thanks

Isaac gollan
workshopninja@gmail.com

Glammies, 2018-12-20:
@vitormhenrique  Sorry I have the same problem. Could you help me?

If i run "sudo python /home/pi/oprint/lib/python2.7/site-packages/octoprint_enclosure/BME280.py 76" i have the positive result. But when i try put the 76 or 0x76 address on octoprint interface not work.

Glammies, 2018-12-20:
problem solved. I remove sudo option on the advanced options.
Thanks

ventilator, 2021-05-20:
> problem solved. I remove sudo option on the advanced options.
> Thanks

same here. BME280 reports 0°C with sudo option, actual temperature without. Thanks for the tipp