vitormhenrique/OctoPrint-Enclosure #140: Neopixel control apparently not working

gertorro, 2018-05-12:
Hi there,
I'm having a rare issue with the enclosure pluging and a strip of neopixel LEDs.
Say I configure GPIO 4 as output (simple toggle on off), with that, I'm able to control pin 7 with no issues. 
Then, I erase that configuration and put instead a Neopixel output on the same physical output and it does nothing when I send a sample color to show.
Any clue why this might be happening? I have attached the configuration just in case it gives you a hint.
Thanks in advance,
Germán

[image]

[image]

vitormhenrique, 2018-05-12:
Yes, I know exactly what is going on, I think I did not fully document on the read-me how to get it working.
You need to use hardware capable PWM pins, and also install some libraries.
Follow this [guide](https://learn.adafruit.com/neopixels-on-raspberry-pi?view=all) install everything and get it working, after that, it should work on the same pin that you configured following the guide.

vitormhenrique, 2018-05-12:
Not really sure, you need to get those libraries installed before getting neopixel to work, post an issue on rpi_ws281x github, they could help you out... https://github.com/jgarff/rpi_ws281x, also try to find other tutorials on installing rpi_ws281x.

gertorro, 2018-05-13:
Hello Victor, in the issues from the WS281X library appears the solution,
apparently the source that hosts the setuptools... file is corrupted.
After deleting it manually and downloading it again from the official python servers it run like a charm.
I leave you the source of the fix: [](https://github.com/jgarff/rpi_ws281x/issues/290)
so you can implement the automatic installation out of the box.
Thanks for the support, I personally think your work is of great value for those with 3D printers! Keep it up!
Regards,
Germán

vitormhenrique, 2018-05-13:
No problem! glad it work out for you!
I had problems trying to install automatically the requirements of this plugin using octoprint pip, I'll try to get this working for following releases.

closed 2018-05-13 by vitormhenrique