vitormhenrique/OctoPrint-Enclosure #141: Possible to use negative from a different source through pins?

areonic, 2018-05-14:
Hello, I am trying to create a status LED bar. Sadly they are 12v, so they cannot be powered through the pi itself. I have a separate power supply for them, but obviously the + and - have to be on the same circuit. Normally I would have just hooked the pins to the board and had enclosure turn on the ground from a pin, but due to voltage, I cant. So what I am wondering is if its possible to run the ground from the power supply into a pin, then when an event or I switch it on, it sends the ground through to another pin that is connected to the - of the LEDs. Just to make it easy, Im using simple numbers instead of right ones, so bear with me......So its an RGB LED strip, the power is straight to the power supply, then the 3 color/grounds to pin 5, 6, 7....Then the ground from the power supply is on pin 10....Then when I switch or an event, it connects pin 10 and 7, to pull the ground from the same power source. Hope this make sense.....thank you

vitormhenrique, 2018-05-14:
ok, it's your writing is very very confusing. 
From what I understand, you have a 12 Volts RGB strip and you want to control with the plugin. Is that correct? 
If so your first assumption is incorrect, negative and positive do not need to be on the same "circuit" as the raspberry pi, you can use "relays", "mosfets", opto-isolators to isolate circuits.

Check this tutorial on how it is possible to control 12V RGB strips using "arduino" : https://learn.adafruit.com/rgb-led-strips?view=all#usage
Also check this one:
https://www.makeuseof.com/tag/connect-led-light-strips-arduino/

You will need 3 PWM outputs to control the intensity of each color, but you will not have a nice color picker to use. 

My recommendation is: buy a new neopixel led strip, those are 5V, you can control directly with the reaspberry pi, the plugin supports it, and you will have control over it with nice color pickers, and even GCODE commands that you can add to your print queue to change colors depending on the print status.

closed 2018-05-14 by vitormhenrique

areonic, 2018-05-15:
Thank you so much for responding. I took your advice about using the pixel type led strips and looked into them. Then noticed I have a whole brand new role from a long time ago, except they weren't called pixel then, so thats probably why I didnt realize, also didnt know they ran on 5V. When I looked in the settings in the enclosure plugin, it has only 1 pin for them though and the set I have has 2....well 4, Pos, Neg, cK, sI. Also, if I understand you correctly, I can plug them directly into the Pi and let it control them. You also said I can make it change its colors based on commands, or gcode....Do you have instructions for that somewhere?

Thank you

vitormhenrique, 2018-05-15:
Neopixel is a generic name for ws2812b led strips like [this](https://www.ebay.com/itm/WS2812B-5050-RGB-LED-Strip-30-60-144-LEDs-M-ws2812-IC-Individual-Addressable-5V/262452005540?hash=item3d1b5bcaa4:m:mkMwXOyKYoE3TxVrA-ee05w),  if your strip has more than 3 pins it is not a neopixel strip and will not work with the plugin.
Read this issue to learn hot to get it working with raspberry pi first:
https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/140
And them check the wiki for more advanced usage: https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/G-CODE-Control

I'm sorry but I can not give you more assistant than pointing the right directions, you need to read a little bit more and get it working before I can help you with the plugin.

areonic, 2018-05-15:
Alright, so I spent all night and morning trying to make a pixel strip using an arduino nano as the microprocessor and these....
https://www.amazon.com/gp/product/B00MIJVKXW/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

and have gotten nowhere. So I was going to try and clear up my original question, since it didnt make sense.

Can I take a ground from an external source and plug it into a pin on the raspberry pi. 
Then take the grounds from the RGB strip, so its 3 grounds, plug them into 3 diff pins.
Then on an event, it opens a path from the external source, to one of the 3 ground pins.

This is just an example......
RGB + --------- External 12v
RGB R---------- Pi  Pin 5
RGB G---------- Pi  Pin 6
RGB B---------- Pi  Pin 13
External Ground-------------Pi  Pin 23    (External ground from 12V source)

Then an event opens a connection from Pin 23 to Pin 5, to give it the ground from the same source of the 12V.

I hope that makes more sense, thank you!  I really am sorry to keep this going, I am really just trying to find a way to get a status bar on my setup with the materials I have. All I want is a bar with 5 LEDs on it to be one color while printing, and one color when finished, that kind of stuff.

vitormhenrique, 2018-05-15:
No, you can NOT do what you are talking, you can NOT use 12V strips directly on raspbery pi or arduino, if you do it you are going to burn your raspberry pi.  I already sent you an link explaining what you need to do to use 12V strips with arduino, read it here: https://www.makeuseof.com/tag/connect-led-light-strips-arduino/

Ws2801 led strips will NOT work with the plugin, only **ws2812b**. BUT, it's a god time for you to learn how they work, follow this [guide](http://www.instructables.com/id/Extreme-Ambilight-System-using-DotStars-Led-ws2801/) and get it working on your arduino nano.

It's NOT something that I recommend for a beginner, just **buy** a strip that I sent and you will get it working faster. You need to be patience and do more research before using any electronics on your pi, it's OK to be a beginner and to ask.