vitormhenrique/OctoPrint-Enclosure #142: Direct NeoPixel don't work with DMA 5 on Rasp 3B

hprotzek, 2018-05-23:
The default setting in `neopixel_direct.py` with DMA 5 doesn't work on Raspberry Pi 3. Test with the `strandtest.py` (default DMA 10) of the rpi_ws281x library were successful.

Also there seem to be an issue with DMA 5: https://github.com/jgarff/rpi_ws281x/issues/224

Can you make the DMA channel configurable? or change the default to 10?

gordonturner, 2018-05-26:
To provide another data point, had the same issue, edited `neopixel_direct.py` to switch the DMA from 5 to 10:

```
vi ./oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py
```
```
LED_INVERT = False
# CHANGED
#LED_DMA = 5
LED_DMA = 10
LED_FREQ_HZ = 800000
```

Also on a Raspberry Pi 3.

vitormhenrique, 2018-05-29:
You might get away removing sudo from from the advanced sections and giving pi user access to gpio, but I don't know...

For now as a work around manually update the DMA from 5 to 10. I'll make it configurable on a patch.

gordonturner, 2018-05-30:
@vitormhenrique tried a couple of time to set the permissions (please see below) but none of them seemed to work.  I would of course prefer not to use sudo if possible :-)

If there is something I am missing, please let me know.

Attempt 1
---------

- Remove sudo requirement

- Reference:

https://raspberrypi.stackexchange.com/questions/40105/access-gpio-pins-without-root-no-access-to-dev-mem-try-running-as-root

- Check that /dev/gpiomem has the correct permissions:

```
ls -l /dev/gpiomem
```
```
crw-rw---- 1 root gpio 248, 0 May 26 15:06 /dev/gpiomem
```

- If not 'crw-rw----' and 'root.gpio', correct permissions:

```
sudo chown root.gpio /dev/gpiomem
sudo chmod g+rw /dev/gpiomem
```

- Confirm group membership for user pi, login as pi and run:

```
groups
```
```
pi adm tty dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi
```

- If not a member of the gpio group:

```
sudo adduser pi gpio
```

- No change

```
Can't open /dev/mem: Permission denied
```

Attempt 2
---------

Open up /dev/mem to gpio group

- Current permissions:

```
ls -lart /dev/mem
```
```
crw-r----- 1 root kmem 1, 1 May 26 15:34 /dev/mem
```

- Change:

```
sudo chown root.gpio /dev/mem
sudo chmod g+rw /dev/mem
```

- Updated permissions:

```
ls -lart /dev/mem
```
```
crw-rw---- 1 root gpio 1, 1 May 26 15:34 /dev/mem
```

- No change

```
Can't open /dev/mem: Permission denied
```

vitormhenrique, 2018-06-14:
Added on the [new release](https://github.com/vitormhenrique/OctoPrint-Enclosure/releases/tag/4.09) update and test it out...

closed 2018-06-14 by vitormhenrique

vitormhenrique, 2018-06-15:
for documentation purposes: dma is set on the advanced section of the plugin, if empty it will use 10 as default.

gordonturner, 2018-06-16:
Thanks @vitormhenrique! I will check it out when I do my Octoprint updates.