vitormhenrique/OctoPrint-Enclosure #143: Problem with ws2812 led strip, directly on the raspberry

thearmm, 2018-05-29:
Hello

I use a WS2812 led strip directly on the raspberry, but it does not turn on. Yet when I turn it on with a python script, it turns on to dismiss the hardware problem.

I do not see where the problem comes from ...

vitormhenrique, 2018-05-29:
without logs, print screens, more information there is nothing I can do to help you... the very first notice on the read-me is:

> Before opening an issue...:
> Check the troubleshooting guide. Issues with no log, no print screen will be closed until the necessary documentation is available.

Your issue might be related with this one https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/142 but who knows?

thearmm, 2018-05-29:
I have no alert message to show and I am a novice, I looked at the issue 142 but nothing changes by changing the DMA to 10. And I also looked at the Troubleshooting Guide..

vitormhenrique, 2018-05-29:
But still, I need the **octoprint log**, **print screens** to see your setup and everything else that you are using, for example, I need the code that you are using to test the neopixel, to compare with the one being used by the plugin.

thearmm, 2018-05-29:
ok, my led strip works with any scripts that neopixel provided as an example

[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2051598/octoprint.log)

vitormhenrique, 2018-06-01:
You need to enable debug information on the advanced section of the plugin. Then try again and send me the log again.
Also, you need to double check the IO number that you are using, by default the plugin uses BCM pin numbering.

Also you configured it to use pin number 21, that only works for PCM mode, the plugin uses PWM mode. 
from the [rpi_ws281x](https://github.com/jgarff/rpi_ws281x) documentation:

> PWM0, which can be set to use GPIOs 12, 18, 40, and 52.
> Only 12 (pin 32) and 18 (pin 12) are available on the B+/2B/3B
> 
> PWM1 which can be set to use GPIOs 13, 19, 41, 45 and 53.
> Only 13 is available on the B+/2B/PiZero/3B, on pin 33

Change it to pin 18 and give it a try...

thearmm, 2018-06-02:
I tried on the pin 18 but it still does not work. Here are the logs
[octoprint (2).log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2065466/octoprint.2.log)

vitormhenrique, 2018-06-03:
And I’m guessing you changes the DMA from 5 to 10 on the plugin script? I noticed that you are using rpi3 B. Have you done this?

thearmm, 2018-06-03:
Yes and yes !

vitormhenrique, 2018-06-03:
Can you give me access to your raspberry pi, I need ssh access, so ip, password, everything (you will need to open ports 22 and 80 on your modem so I can remote access it)... send on my email that is linked here. Other people have been using this plugin with ws2812 strip. So I'm not really sure.
Also point me to the location of the test script that is working...

thearmm, 2018-06-03:
The problem is that I am in a residence and the ports are blocked .... But I hamachi to install on it if you want

vitormhenrique, 2018-06-03:
That would work, send me all the login information by email, but first can you explain how did you change the DMA from 5 to 10 on the plugin?

vitormhenrique, 2018-06-03:
ok, you said that you got it working on ther examples right?

So let's do this, create a new script, called **test.py** with the following code:

<pre><code>from neopixel import *
import sys
import time

LED_INVERT = False
LED_DMA = 10
LED_FREQ_HZ = 800000

if len(sys.argv) == 8:
    LED_PIN = int(sys.argv[1])
    LED_COUNT = int(sys.argv[2])
    LED_BRIGHTNESS = int(sys.argv[3])
    red = int(sys.argv[4])
    green = int(sys.argv[5])
    blue = int(sys.argv[6])
    address = int(sys.argv[7], 16)
else:
    print("fail")
    sys.exit(1)

strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT)
strip.begin()

color = Color(green, red, blue)

for i in range(LED_COUNT):
    strip.setPixelColor(i, color)

strip.show()

print("ok")</code></pre>

Try to run it by using:

`sudo python test.py 18 60 255 255 255 255 0`

This is the exact same code that the plugin is using, the only difference is that I changed the DMA form 5 to 10.

thearmm, 2018-06-04:
It's Working !

vitormhenrique, 2018-06-04:
What does that mean? Can you explain how you changed the DMA on the original plugin?

vitormhenrique, 2018-06-04:
You need to improve your communication, I understand that English is not your first language, but I ask a lot of questions and you only answer "**yes**" "**no**" I want explanations.
I asked multiple times how have you changed that original DMA channel, to test for the issue https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/142, where did you find the script on the enclosure plugin?
If you don't answer my questions I'll give up helping you and close this issue.

thearmm, 2018-06-04:
On the original plugin I changed the dma to the following address:

./oprint/lib/python2.7/site-packages/octoprint_enclosure/neopixel_direct.py

And sorry for my english
[image]

vitormhenrique, 2018-06-04:
No worries about the English, you just need to help me help you...
What did you meant by "**But that planted the Raspberry**"?

thearmm, 2018-06-04:
The Rasp did not answer anymore. I had to restart it

thearmm, 2018-06-04:
I tried with the pin 12 and 18. With more or less led and different color but nothing, it should work because it's the same code
[image]
[image]

thearmm, 2018-06-04:
I tried with the enable and disable command but not anything

vitormhenrique, 2018-06-04:
the only thing I can do is to connect on your pi, and you having a webcam so I can see if the neopixel is working...
Plenty other users have it working, You said that examples worked fine from adafruit tutorial, but my simple script froze your raspberry pi.  I suggest downloading a clean image an installing everything from scratch, either way there is nothing else I can do if I can't connect to your raspberry pi.

You need to get the following script working reliable on your pi, test different DMA, different pin, also change the brightness, if you are using the same power supply for neopixel and your rpi try using a different one, try using level sifters for better communication with the strip...

<pre><code>from neopixel import *
import sys
import time

LED_INVERT = False
LED_DMA = 10
LED_FREQ_HZ = 800000

if len(sys.argv) == 8:
    LED_PIN = int(sys.argv[1])
    LED_COUNT = int(sys.argv[2])
    LED_BRIGHTNESS = int(sys.argv[3])
    red = int(sys.argv[4])
    green = int(sys.argv[5])
    blue = int(sys.argv[6])
    address = int(sys.argv[7], 16)
else:
    print("fail")
    sys.exit(1)

strip = Adafruit_NeoPixel(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT)
strip.begin()

color = Color(green, red, blue)

for i in range(LED_COUNT):
    strip.setPixelColor(i, color)

strip.show()

print("ok")</code></pre>

Again, the only thing I can do now is if you send me an email and instructions to access you pi, I'm going to close this issue as I'm 100% sure that is not a plugin bug.

closed 2018-06-04 by vitormhenrique

vitormhenrique, 2018-08-06:
@thearmm you might try the changes that I made on https://github.com/vitormhenrique/OctoPrint-Enclosure/issues/172