vitormhenrique/OctoPrint-Enclosure #144: Duty Cycle Error With PWM

mfocht001, 2018-06-01:
First, thanks for your efforts. This is a really cool plugin!

Now the issue:

I have created a Raspberry Pi Output in the Enclosure Plugin for an  **Output Type** of PWM button and I am getting a duty cycle error message when pressing button in UI. Also, for PWM Output type does not display or has no option to view in UI. 

I have tried the following:

1. Followed troubleshooting guide and removed all configured outputs (no inputs used).
2. Save blank configurations and rebooted RPI.
3. Save blank configurations and restarted Octo server.
4. Validated pins are correct. Setup an output using **Regular IO** and button works and lights up LED.
5. Checked config.yaml and the values are being saved under the plugin tree.

Hardware Raspberry Pi 3 PWM pin 18

Screenshots

[image]

vitormhenrique, 2018-06-01:
Hello, what do you mean by "PWM Output type does not display or has no option to view in UI." ?
Your first print screen show the UI that it is supposed to show. What happens if you type a number, between 0 to 100 on the label that has the "off" grayed out and press the marker?

mfocht001, 2018-06-01:
I mean that the PWM Output Type does not show like the IO Output Type.

Like this:
[image]

When you create PWM Output Type it only does this:
[image]
 
PWM Object Type does not show this option:
[image]

If I try to add a value in the "off" field, it works but not unless I set it there. I am trying to simply add a button like the IO type and have three options show. One for low duty cycle/Medium/High.

mfocht001, 2018-06-01:
Trying to do this but with PWM to dim lights:

[image]

vitormhenrique, 2018-06-02:
That is not suported. You can only control pwm using the ui section on the enclosure tab, you can also control it using custom g-code commands or link it to a temperature sensor.

vitormhenrique, 2018-06-02:
Actually I figure it out a way that you can do it. 
Enagle g-code commands over the advanced section of the plugin, set up one PWM output. 
Than you set up three g-code outputs, and send different g-codes that will control your “pre sets pwm values”.
Look the wiki on how to set up g-cose outputs.

closed 2018-06-02 by vitormhenrique

mfocht001, 2018-06-02:
Thats only half of it. I am pointing out two things. One is what I was trying to do. which I get that it's not supported but there is still a problem that PWM Output type. It ignores the PWM values you enter and does not use the default_duty_cycle,

vitormhenrique, 2018-06-02:
No no, you did not read the documentation and how it is supposed to work.
The default value is going to be written automatically if and only if you choose “automatically start up” for that output.
You can’t just press the button and expedt it to write it. If you choose to automatically start up that output, when a print starts it will automatically set that value for that output. Manually you can overwrite that value using the UI, set ir over the UI at any time or use the g-code commands.
If you just press the button on the output it will complain and expext you to tell it what value to write. This is by design.
If you want “pre set” values the only way is to use “g-code” outputs, you can creat as many as you want, link for that pwm output and it will create a button on the UI. When you press it it will “write” a PWM value that configure on the g-code.

vitormhenrique, 2018-06-02:
The only way that it is supposed to work is by specifically writing the PWM value on the enclosure tab. And you already said that it was working doing this.
What I'm telling is that you can however accomplish what you want, that is creating buttons to set the pwm value to a specific value, even a "default" value.  I'll improve the documentation to avoid confusion, but this is only way that the plugin was designed.