vitormhenrique/OctoPrint-Enclosure #145: 1 GPI Input should 2 trigger 2 events

Fox2525, 2018-06-03:
Hello I am using Octoprint with the enclosure plugin. 

The enclosure plugin is on 4.0.7. I have a smoke detector attached on pin 17 on my raspberry. When the smoke detector is triggered by an alarm it should trigger my powerswitch tail 2 and my second beeper module. 

(The smoke detector is in the enclosure, I use the beeper wire of the the smoke detector. The smoke detector however isnt that loud anymore because I use the beeper current for the trigger action). Thats why I use a second beeper module that sits not in the enclosure to amplify the beeper sound). 

Above 4.0.0 it is not possible for me to trigger both events anymore. I can only trigger one event. So I cannot use pin 17 to say for example if Pin 17 is on ground then trigger the powerswitch and the beeper module even if I set both conditions. before 4.0.0 it was possible for me to trigger both events when one conditions is met. 

I hope you can help me and I hope you understand my problem. Anyway the plugin is great :). 

Maybe you have also an option to downgrade to an older version ??

vitormhenrique, 2018-06-04:
I'm playing around with the code to improve this, it is possible to use one "trigger" to control two outputs, but it is only possible to add one "event" detect for each GPIO, so it is not possible, for example to have one GPIO event on the rise, and another on the fall for the same GPIO. I'll try to improve this and test it out....

Fox2525, 2018-06-04:
Thanks for your reply and thanks for the link with the transistor. The smoke detector should trigger the event not the raspberry so I could use the transistor base for the beep module and the emitter and collector for the 3,3V of the raspberry to trigger a gpio and that triggers then the powerswitch tail 2 to switch of. So in theory the small current of the base of the transistor will not draw enough current to limit the sound of the beeper module. Does the transistor base need less current than the GPIO of the raspberry ? Otherwise I would have the problem with the low noise in this setup too . hmm what do you think.

vitormhenrique, 2018-06-04:
I got totally lost on your comment, lol...
Can you draw a schematics on what you want to do?
But never the less, I'm updating the plugin so you will be able to setup multiple triggered events using the same "input". I'll probably need few days to test and commit the changes.

vitormhenrique, 2018-06-14:
Added on the (new release)[https://github.com/vitormhenrique/OctoPrint-Enclosure/releases/tag/4.09] update and test it out...

closed 2018-06-14 by vitormhenrique