vitormhenrique/OctoPrint-Enclosure #146: DHT 11

mserowik, 2018-06-04:
i am having a issue with the enclosure.  
I have a DHT 11 attached to the pi and configured it in the enclosure plugin.
When I use putty to test the sensor it work fine.
sudo ./AdafruitDHT.py 11 20

when i try to see the setting in the plugin they are blank

any ideas on how to troubleshoot this?
thanks

vitormhenrique, 2018-06-04:
There are a lot of information on previous issues, on the troubleshooting guide and etc...
If you don't enable debug information, and share octoprint log I can not help you.

DHT sensors are not recommended, they are not reliable, and some times they will need minutes to start working, but again, without print screens and logs there is nothing I can do to help you....

mserowik, 2018-06-04:
> Check this, this and this.

The first link solved it.  I deselected the sudo in the advance option now all works

Thanks for quick answer and a useful plugin!!!!

Michael A. Serowik

vitormhenrique, 2018-06-04:
No worries....

closed 2018-06-04 by vitormhenrique