vitormhenrique/OctoPrint-Enclosure #148: Auto shutdown when printing is canceled

daved3luxe, 2018-06-12:
The Temperature / Humidity Control Auto Shutdown will not work if you cancel a printjob or the printer disconnects. Auto Shutdown is activated. Can you adjust that?

vitormhenrique, 2018-06-12:
That is by design, when the printer is cancel it will run the scripts to automatically "turn off" all the gpios that you have marked.  Many users use it like that, if you have fans, lights, and the printer fail, cancel, they will turn those off.
If you want to have control when the auto-shutdown is activated that is a feature enhancement. And as I said previously I'm not changing or adding features for the plugin on a near feature, last release took me 2 months of coding and I'm working on new projects now, I'll consider for next release, but I it will not be on my top priority.

vitormhenrique, 2018-06-12:
Please confirm if my comment is what you expected and I'll update this to a feature enhancement.

vitormhenrique, 2018-06-14:
@daved3luxe could you please confirm that what you expected is what I described on my initial comment?

daved3luxe, 2018-06-14:
Sorry, Yes that is what I mean.

vitormhenrique, 2018-06-14:
I'll change it for a feature enhancement.

vitormhenrique, 2018-06-14:
Added on https://github.com/vitormhenrique/OctoPrint-Enclosure/releases/tag/4.09

closed 2018-06-14 by vitormhenrique