vitormhenrique/OctoPrint-Enclosure #149: [Request] Start Print / Cancel Print toggle mode

CapnBry, 2018-06-15:
I thought it would be fun to add some big ole arcade-sized buttons to my printer for quick Start Print or Cancel buttons and was super pleased to find there was already a plugin for this and that it is so dang configurable it can do almost anything!

I was hoping to just have one button that both starts the print and cancels the print, a sort of start/stop button. There's a "Cancel" printer action and a "Toggle" action, but the toggle pauses/resumes and I was hoping for start / stop. Usually this is because the beginning of the print is messing up and I need to restart it. I notice there's no "Start Print" action, probably because there's no way to select what file to print, but that's fine by me because I just want to start the currently selected print. Is there a way to add "togglejob" mode or something to [the other commands](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/octoprint_enclosure/__init__.py#L1076) something like:
```
                    elif rpi_input['printer_action'] == 'toggle' or rpi_input['printer_action'] == 'togglejob' :
                        self._logger.info("Printer action " + rpi_input['printer_action'])
                        if self._printer.is_operational():
                            if self._printer.is_printing():
                                if rpi_input['printer_action'] == 'toggle':
                                    self._printer.pause_print()
                                else
                                    self._printer.cancel_print()
                            elif self._printer.is_paused():
                                self._printer.resume_print()
                            elif self._printer.is_ready():
                                self._printer.start_print()
                        else:
                            self._printer.connect()
```
or just a whole new code block instead of editing 'toggle'
```
                    elif rpi_input['printer_action'] == 'togglejob':
                        self._logger.info("Printer action togglejob.")
                        if self._printer.is_operational():
                            if self._printer.is_printing():
                                self._printer.cancel_print()
                            elif self._printer.is_ready():
                                self._printer.start_print()
                        else:
                            self._printer.connect()
```

Also I'm not sure if it was added after you wrote the toggle code, but printer also now has `toggle_pause_print()` which could reduce the is_printing/is_paused code of 'toggle' to just a single line.

vitormhenrique, 2018-06-15:
Pretty easy to implement,  But I don't think we need to create a new type _togglejob_, I can just test if it's printing, or not, it's ready or not and figure it out if should start / stop or pause a print job.
And yes, I haven't implemented the start because currently (from the plugin) there is no way to select the file, maybe someday when I implement LCD / OLED display I will have that sorted out...

vitormhenrique, 2018-06-15:
But for now I can implement "if the file is loaded" start the print...

CapnBry, 2018-06-15:
That would be great but how can it tell if you want to pause/resume vs cancel a print? I would think pause and cancel would have the same `printer` starting state, `is_printing()`. I'm not as familiar with the API though so I defer to your expertise 😃

vitormhenrique, 2018-06-15:
You were right, I do need to keep different toggles...
I made the changes on the dev branch and added a new FR that was here for a while, adding the possibility to execute any shell script as an output. I have not tested it yet because I don't even have a rpi with me right now, it might be possible (very likely)  that the code won't even "compile"..  
I'll test this weekend and if everything is alright I'll put it production...

vitormhenrique, 2018-06-20:
Implemented on new release.

closed 2018-06-20 by vitormhenrique