vitormhenrique/OctoPrint-Enclosure #150: Temp controlled output does not obey delay after print finishes

bumbalu, 2018-06-15:
My enclosure ventilation fan is configured as "temperature controlled" , auto start up (when print starts) = yes, startup delay =0, auto shutdown = yes, shutdown delay = 600, temp control type = cooler, default value = 56, deadband = 2. 

When print finishes the fan goes to "off" and does not ventilate/cool the enclosure. But it should. 

Which log files should I provide to find out the cause?
And again, thank you for the best plugin ever, Vitor.

vitormhenrique, 2018-06-15:
I was reviewing the code, I have not implemented the delay when the output type is set to "Temperature / Humidity Control", so it was really a bug letting you choose the delay. But I think I'll implement as it might be a good feature.

bumbalu, 2018-06-17:
Thank you. It is be a good feature when you print toxic abs or nylon and want to ventilate the enclosure for like 10 minutes after print finishes. I would wait for this implementation :-)
thank you!

ShagoY, 2020-01-30:
> Fixed on new release, but I'm actually have a better solution for this on a few days.

What is the better solution for this ?

> You can create a pwm linked temperature with a default off 100% and auto shutdown. This will keep a fan slower and controling cooling the temperature while printing, set it to 100% when the print is done and turn it off with the delay.

Would it be easier to leave the pwm linked temperature script constantly active ?