vitormhenrique/OctoPrint-Enclosure #152: Temperature sensor returns 0c

SMenn, 2018-06-18:
**Describe the bug**
Hi,
I am using the enclosure plugin, with a AM2302 sensor from amazon. (https://www.amazon.com/gp/product/B014SMNBJC/ref=oh_aui_detailpage_o01_s01?ie=UTF8&psc=1)

I followed the documentation on github, however my temp and humidity readings in OctoPrint are both either 0 or nonexistent. I am using GPIO Pin 4, and if I run `sudo ./AdafruitDHT.py 2302 4` it returns correct values in SSH. I have tried un-checking "use sudo commands".

**Expected behavior**
Accurate Temp and Humidity readings,

**Additional context**
[image]
[image]
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2112506/octoprint.log)

vitormhenrique, 2018-06-18:
I found a small issue on the code, but I don't think it is related with your issue. I have told time and time again that dht sensors, including the "AM2302 " are not reliable under linux, they rely on real critical timing and sometimes linux will fail for minutes and minutes to read it, specially when trying to run the script every 10 seconds.
Your log had issues when using the sudo, after that I don't see any other issue related with the temperature sensor, I'm fixing the issue that on another method now, uninstall and install the plugin again, you also need to enable debug info and the temperature log (after you enable the debug) so I can see more info on your system.

vitormhenrique, 2018-06-18:
Uninstall the plugin and install it using the development branch, I fixed an error on another method, https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip 
There are stuff on the dev branch that I haven't tested (new features) but it should be fine.
Enable the logs that i told, save the settings and send me again.

SMenn, 2018-06-18:
Issue is fixed after re-install. Thanks Vitor!

vitormhenrique, 2018-06-18:
reinstalled using the dev?

vitormhenrique, 2018-06-18:
let me know if you are using the dev branch

closed 2018-06-18 by vitormhenrique

SMenn, 2018-06-19:
I used Dev branch and it fixed the issue

evanquinnalaska, 2018-07-02:
I have the same issue as described in the original post.  Installed your https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip to see if it would also fix my occurrence of this issue but it did not. 
Pi3b, Current octoprint, AM2302 Sensor. Dev build provided above in this thread as current installed Plugin. Settings below in images to hopefully help with ironing out the issue. If there is any other feed back I can provided please let me know how I can provide it :)
Example of it working from SSH
[image]
Input settings to read from GPIO3
[image]
Plugin Displaying 0 degrees on Dev build of Enclosure plugin. 
[image]

evanquinnalaska, 2018-07-02:
I turned off sudo after learning how to find the logs and that seemed to resolve the issue.

vitormhenrique, 2018-07-02:
@evanquinnalaska new images of octoprint apparently don't like the sudo command.