vitormhenrique/OctoPrint-Enclosure #155: IFTTT API KEY doesn't get saved

wilhelmzeuschner, 2018-06-19:
Description and reproduction:
Whenever I copy and paste my Key into the field, it gets "magically" removed from it when hitting the "Save" button. Therefore no notifications can be received.

The same thing happens when I type random charters, which I did in my recording.

Due to the lack of a proper screen-recorder I used the "Steps Recorder" provided by the Windows OS / Microsoft.
I have attached the file below, it opens in Internet Explorer.
[Key_problem.zip](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2116923/Key_problem.zip)

vitormhenrique, 2018-06-19:
Yeap I think the issue was introduced not really sure when, I think I fixed on the dev branch.

Uninstall the plugin and install it using the development branch, I fixed an error on another method, https://github.com/vitormhenrique/OctoPrint-Enclosure/archive/dev.zip
There are stuff on the dev branch that I haven't tested (new features) but it should be fine.

I have not pushed to production because I need to test it better...

vitormhenrique, 2018-06-20:
Fixed on new release.

closed 2018-06-20 by vitormhenrique