vitormhenrique/OctoPrint-Enclosure #158: Upgrading deleted all configuration settings

calston, 2018-06-20:
Did the auto upgrade to 4.10 and I lost all my settings. 

SAD FACE

vitormhenrique, 2018-06-20:
What version you had before?

vitormhenrique, 2018-06-20:
If you had a newer version of the plugin I need to check this, if you had an old version than it was documented clearly on the [read-me](https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/master/README.md#before-opening-an-issue)

From versions <4 to newer I changed completely the architecture of the plugin and needed to redo all logic, implemented a bunch of new features so old settings were simply to difficult to migrate.

vitormhenrique, 2018-06-20:
ops... i think I forgot a line "self._settings.set(["rpi_outputs"], old_outputs)" on the settings migrate :-1:

vitormhenrique, 2018-06-20:
it should be fixed on 4.11 but now that settings are gone, they are gone...

But let me know from which version you were upgrading from.

closed 2018-06-20 by vitormhenrique

calston, 2018-06-20:
Yeah it was definitely a recent version, used my printer about 2 weeks ago.

They're goooone,, so gooone, so sad, too bad. If only I had written down which GPIO I connected. 

thanks :)

vitormhenrique, 2018-06-20:
did you have inputs as well or just outputs? yeah, i know, it's a pain, if you have log enable you might be able to at least see the pins that were connected.

calston, 2018-06-20:
I just have outputs and an 18b20 temperature sensor which was on the default pin