vitormhenrique/OctoPrint-Enclosure #160: Temperature Control - I dont get it

echloton, 2018-06-21:
Hey there,
im having some trouble setting up the temp. control.
What i want: If the temp is above 30°C set GPIO 21 HIGH, if under 30 GPIO 21 LOW.

If i manually trigger GPIO 21 it works. I tried Values at Deadband (0-30/30/30-100) but none worked. Temp-Sensor is working and showing Temps (that make sense)

Im quite sure i get something wrong here and theres no bug.
[image]
[image]
[octoprint.log](https://github.com/vitormhenrique/OctoPrint-Enclosure/files/2124955/octoprint.log)

Best regards

vitormhenrique, 2018-06-21:
dead band is just a integer, a value, not a range.
https://en.wikipedia.org/wiki/Deadband

you need to set deadband to zero to get what you want.

closed 2018-06-21 by vitormhenrique

echloton, 2018-06-21:
Ok, i have tried that. But Where do i Set the temp at which the pin goes high/low?

Cheers

vitormhenrique, 2018-06-21:
on the enclosure tab, it's going to have a "set temperature" on the grid that displays the temperature.
you can also make it auto turn on and off and set a "default" value.

echloton, 2018-06-21:
Ill try that, thanks!

vitormhenrique, 2018-06-21:
Enclosure tab:
[image]

If you want to automatically start cooling with the print:
[image]

echloton, 2018-06-21:
Thx alot, worked perfectly! Or works..better.