vitormhenrique/OctoPrint-Enclosure #164: Gcode to set heater temp

SMenn, 2018-07-02:
Hi Vitor,

I'd love to be able to have the ability to set the target ambient temp using custom Gcode. I print in many different materials, and It'd be very helpful to put a line in my starting Gcode for each one instead of manually setting the temp.

vitormhenrique, 2018-07-02:
This feature is already implemented, you just need to enable g-code control under the advanced section and use the correct g-code commands:
https://github.com/vitormhenrique/OctoPrint-Enclosure/wiki/G-CODE-Control

So for example, let's say that you output for the heater has ID number 5 you could send `ENC O5 W0 S40` To set the temperature to 40 and start printing right away, or sending `ENC O5 W1 S40` would set the temperature, wait the temperature to reach 40 and then start printing.

vitormhenrique, 2018-07-02:
by the way, you can use the same gcodes as new outputs and create UI buttons for automated tasks, and link to physical buttons for more automations, so you can have buttons on the enclosure to set temperatures, extrude filaments, move extruder around etc...

closed 2018-07-02 by vitormhenrique